pybamm-team/PyBaMM #2964: [Bug]: CasadiSolver._integrate(): is increment of "count" misaligned?

ejfdickinson, 2023-05-17:
### PyBaMM Version

23.4

### Python Version

3.9

### Describe the bug

Is the increment of `count` misaligned in `CasadiSolver._integrate()` (code snippet below)?

From the documentation, `max_step_decrease_count` is the number of allowed times the maximum step can decrease. This decrease occurs at the line `dt /= 2`.

But `count += 1` is aligned in the main step loop outside the except block, so it will be incremented once per window, whether or not the `try` block succeeds or `dt` has been reduced.

Should `count` simply be incremented adjacent to the code that changes `dt`?

`while t < t_f:
                # Step
                solved = False
                count = 0
                dt = dt_max
                while not solved:
                    # ...
                    try:
                        # ...
                    except pybamm.SolverError as error:
                        pybamm.logger.debug("Failed, halving step size")
                        dt /= 2
                        # ...
                        if count >= self.max_step_decrease_count:
                            message = (
                                "Maximum number of decreased steps occurred at "
                                f"t={t} (final SolverError: '{error}'). "
                                "For a full solution try reducing dt_max (currently, "
                                f"dt_max={dt_max}) and/or reducing the size of the "
                                "time steps or period of the experiment."
                            )
                           # ...
                    count += 1
`

### Steps to Reproduce

n/a (bug spotted in code review)

Could reproduce with any model with events requiring a substantial decrease in `dt_max` - does the error condition come too early?

ejfdickinson, 2023-05-17:
Off-by-one still gives unspecified behaviour if `self.max_step_decrease_count == 0`, right?

My main concern was that subsequent code changes could very easily introduce more marked errors here, if the program flow changes - so flagging for your attention. Even if 'bug' is an overstatement now, it looks worth fixing: if what `count` is meant to be counting is changes to `dt`, put the `count` increment in the same block as the `dt` change so you can't have one without the other.

tinosulzer, 2023-05-17:
Got it. Seems like an easy and not urgent fix so I'll leave it for someone looking for a good first issue

closed 2023-05-30 by rtimms