pybamm-team/PyBaMM #2975: error parameter

mooncode30, 2023-05-22:
input:
#Calculate stoichiometries at 100% SOC
esoh_model = pybamm.lithium_ion.ElectrodeSOH()
esoh_sim = pybamm.Simulation(esoh_model, parameter_values=parameters)
param = DFN_21700.param

output :
AttributeError: module 'pybamm.models.full_battery_models.lithium_ion' has no attribute 'ElectrodeSOH'. Did you mean: 'electrode_soh'?

after i change it with 'electrode_soh' the problem is

input :
esoh_model = pybamm.lithium_ion.electrode_soh()
esoh_sim = pybamm.Simulation(esoh_model, parameter_values=parameters)

output:
esoh_model = pybamm.lithium_ion.electrode_soh()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'module' object is not callable

then i try to calculate stoichiometries at 100% SOC with 
input:
parameters.set_initial_stoichiometries(1);
param = DFN_21700.param

# Update parameter values with initial conditions
c_n_max = parameters.evaluate(param.n.c_max)
c_p_max = parameters.evaluate(param.p.c_max)
parameters.update(
    {
        "Initial concentration in negative electrode [mol.m-3]": esoh_sol["x_100"].data[0] * c_n_max,
        "Initial concentration in positive electrode [mol.m-3]": esoh_sol["y_100"].data[0] * c_p_max,
    }
)

output:
[image]

please i need help about this problem

mooncode30, 2023-05-22:
output:
[image]

can you correct the code please?
i don't understand the replacement of "esoh_sol" because of from the code 
#Calculate stoichiometries at 100% SOC
esoh_model = pybamm.lithium_ion.ElectrodeSOH()
esoh_sim = pybamm.Simulation(esoh_model, parameter_values=parameters)
param = DFN_21700. param

Vmin = parameters['Lower voltage cut-off [V]']
Vmax = parameters['Upper voltage cut-off [V]']
Cn = parameters. evaluate(param. n. cap_init)
Cp = parameters. evaluate(param. p. cap_init)
n_Li_init = parameters. evaluate(param. n_Li_particles_init)

esoh_sol = esoh_sim. solve(
     [0],
     inputs={"V_min": Vmin, "V_max": Vmax, "C_n": Cn, "C_p": Cp, "n_Li": n_Li_init},
)
print(f"Initial negative electrode SOC: {esoh_sol['x_100'].data[0]:.3f}")
print(f"Initial positive electrode SOC: {esoh_sol['y_100'].data[0]:.3f}")

I have changed it to "parameters.set_initial_stoichiometries(1);"

tinosulzer, 2023-05-22:
I'm sorry, these are very basic Python questions rather than specific Python questions, and we are unable to provide the level of support you are looking for. I would suggest taking a course on Python programming and making sure you are following the latest versions of the PyBaMM examples.

mooncode30, 2023-05-22:
okay i have been trying again. I wanna ask why  x_100 = sol['x_100'].data[0] error? 
the output is :
KeyError: "'x_100' not found. Best matches are []"