pybamm-team/PyBaMM #2979: Implement `gradient_squared` for `FiniteVolume`

brosaplanella, 2023-05-23:
### Description

At the moment `FiniteVolume` does not have a `gradient_squared` method, so it needs to be computed as the inner product of the gradient with itself.

### Motivation

Implementing it would allow some code simplifications in the heat sources (Ohmic & heat of mixing).

### Additional context

Main issue is to find how we want to implement it (mathematically). If the discretised gradient of `u` is `L u`, then the gradient squared is just `u^T L^T L u`, where superscript `T` means transpose. Is this too naive?

arunavabasu-03, 2023-09-08:
@brosaplanella can you assign it to me ?

medha-14, 2024-10-22:
I have tried to implemented the gradient square using the previously discussed method. I am open to exploring other implementations as well. If you have any suggestions.

```
def gradient_squared(self, symbol, discretised_symbol, boundary_conditions):
        """
        Compute the square of the gradient of a symbol.
        
        Parameters
        ----------
        symbol : :class:`pybamm.Symbol`
            The symbol for which to compute the gradient squared.
        discretised_symbol : :class:`pybamm.Vector`
            The discretised variable for which to compute the gradient squared.
        boundary_conditions : dict
            Boundary conditions for the symbol.

        Returns
        -------
        :class:`pybamm.Vector`
            The gradient squared of the symbol.
        """
        # Compute the gradient of the symbol
        gradient = self.gradient(symbol, discretised_symbol, boundary_conditions)
        
        # Get the gradient matrix
        domain = symbol.domain
        gradient_matrix = self.gradient_matrix(domain, symbol.domains)
        
        # Compute gradient squared: (∇u)^2 = u^T L^T L u
        gradient_squared_matrix = gradient_matrix.T @ gradient_matrix
        gradient_squared_result = gradient_squared_matrix @ discretised_symbol
        
        return gradient_squared_result
```

brosaplanella, 2024-10-22:
Hi! Could you please open a PR? It will be easier to discuss there as we will be able to see all the changes & tests.