pybamm-team/PyBaMM #2982: [Bug]: Error when pickling simulation

brosaplanella, 2023-05-24:
### PyBaMM Version

23.4.1

### Python Version

3.9.16

### Describe the bug

When trying to pickle a simulation with experiment I get an error related to a `SwigPy` object that can't be pickled. I haven't been able to trace the origin of such object.

### Steps to Reproduce

Run
```python3
import pybamm
model = pybamm.lithium_ion.SPMe()

experiment = pybamm.Experiment(
    [
        (
            "Discharge at 1C until 2.5 V",
            "Charge at C/3 until 4.2 V",
            "Hold at 4.2 V until C/20",
        )
    ]
)

sim = pybamm.Simulation(
    model,
    experiment=experiment,
)
sim.solve()
sim.save("test.pkl")
```

If instead you replace the last few lines by
```python3
sim = pybamm.Simulation(
    model,
)
sim.solve([0, 3600])
sim.save("test.pkl")
```
it works fine.

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/brosaplanella/support-pybamm/mwe_swigpy.py", line 19, in <module>
    sim.save("test.pkl")
  File "/home/brosaplanella/support-pybamm/env/lib/python3.9/site-packages/pybamm/simulation.py", line 1096, in save
    pickle.dump(self, f, pickle.HIGHEST_PROTOCOL)
TypeError: cannot pickle 'SwigPyObject' object
```

tinosulzer, 2023-05-24:
Pickling simulations is a pain and simulations are super cheap to set up. Do we want to continue allowing this, or suggest that people set up configuration files instead?

tinosulzer, 2023-05-24:
This also kind of relates to #2787

brosaplanella, 2023-05-25:
I think that some optimisers pickle (or do something similar to pickling) when running in parallel. It was not where this issues came up, but had similar issues in the past with pybamm-param. I think for that it would be enough to be able to access an unsolved version.

brosaplanella, 2023-06-05:
Ok, I found the issue. It is with `op_conds_to_build_solvers` which keeps a copy of the solver which is not processed for pickling. It should be an easy fix, we only need to process the solvers in `op_conds_to_build_solvers` similarly as we do in
https://github.com/pybamm-team/PyBaMM/blob/a1e64c7715723b67928a9423e714689d9ee06e83/pybamm/simulation.py#L1090

Note that this issue appears only after merging #2960

closed 2023-06-10 by tinosulzer