pybamm-team/PyBaMM #3003: Improvements to `latexify`

brosaplanella, 2023-05-31:
A few improvements to `latexify`, mostly motivated by printing the SEI models.

- [ ] Integrals currently displays `xn` as integration variable.
- [ ] Definite integrals are not taken into account.
- [ ] Add `SEI`, `inner` and `outer` subscripts.
- [ ] Use Roman `d` in derivatives (rather than maths).

Feel free to extend this list.

fred4ets, 2023-06-13:
@brosaplanella If you're OK with this, I can try to write a patch.

brosaplanella, 2023-06-14:
I am not sure why we chose `standalone` originally (@tinosulzer, @rtimms  and @priyanshuone6, do you remember?). Standalone is useful for small systems, so maybe we could add an extra keyword argument to choose which latex class we want (either standalone or article).

In any case, if we choose article we should definitely change the margins as the default ones are incredibly large.

shibo911, 2025-12-24:
Hi @brosaplanella @agriyakhetarpal @Saransh-cpp, please assign this issue to me.

I have an implementation that addresses the items discussed:

- Derivatives now use roman d (and switch to partials where appropriate).
- Integrals print the correct integration variable and include definite bounds derived from the domain.
- SEI subscripts render as roman “SEI” in LaTeX.
- Added configurability for the output document: you can choose document_class="article" and pass geometry_options (defaulting to landscape,hmargin=0.5cm). This is applied for .tex output and for .pdf via the LaTeX preamble, improving readability and avoiding tiny fonts.

Notes:

- Image outputs (e.g., png/jpg) currently still rely on SymPy’s default standalone class; I can extend the preamble handling to those formats as well if preferred.
- Unit tests for latexify were updated and pass (with the SymPy preview test appropriately skipped on non-Linux platforms).

I’m happy to open a PR with these changes.