pybamm-team/PyBaMM #3011: Interpolant function gives only initial value

chetandharmadhikari, 2023-06-03:
After running the model, the current remains constant = 1st data point. 
The interpolant function does not update the current field. 

However, when the interpolant function is tested in stand-alone, it works correctly. 
It looks like the interpolant function is not called every time-step or does not update correctly. 

Code (Same as the example shown in Loading in current data)
#
# Simulate drive cycle loaded from csv file
#
import pandas as pd    # needed to read the csv data file
import pybamm
# Import drive cycle from file
drive_cycle = pd.read_csv("C:/Users/cheta/Battery/Experimental_Data/US06.csv", comment="#", header=None).to_numpy()

# Create interpolant
current_interpolant = pybamm.Interpolant(drive_cycle[:, 0], drive_cycle[:, 1], pybamm.t)

# Set drive cycle
parameter_values=pybamm.ParameterValues("Chen2020")
parameter_values["Current function [A]"] = current_interpolant

model = pybamm.lithium_ion.SPMe()
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve()
sim.plot(["Current [A]", "Battery vol_tage [V]"])_

brosaplanella, 2023-06-05:
Does anything change if you do `sim.solve([0, 3600])`?

chetandharmadhikari, 2023-06-07:
Hi,

Thank you for getting back to me. I tried to solve with
sim.solve([0,3600]), however, the current value is still constant (first
value from the profile)

[image: image.png]

#
# Simulate drive cycle loaded from csv file
#
import pandas as pd    # needed to read the csv data file
import pybamm
# Import drive cycle from file
drive_cycle = pd.read_csv(
"C:/Users/cheta/Battery/Experimental_Data/US06.csv", comment="#", header=
None).to_numpy()

# Create interpolant
current_interpolant = pybamm.Interpolant(drive_cycle[:, 0], drive_cycle[:, 1],
pybamm.t)

# Set drive cycle
parameter_values=pybamm.ParameterValues("Chen2020")
parameter_values["Current function [A]"] = current_interpolant

model = pybamm.lithium_ion.SPMe()
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve([0,3600])
sim.plot(["Current [A]", "Battery voltage [V]"])

Thanks,
Chetan

chetandharmadhikari, 2023-06-09:
> I can't tell what the problem is. Could you please produce a minimum
> working example
> <https://en.wikipedia.org/wiki/Minimal_reproducible_example> that we can
> run on our machines and debug?

Hello,

Please find below the minimum working example for this code. You can notice
that the current does not change even when provided as a profile.
Also attached same code

#
# Simulate variable current profile
#
import numpy as np
import pybamm
current_profile = np.array([1.0,2,3,4,3,2,1])
time_profile = np.array([0.0,10.0,20,30,40,50,60])

current_interpolant = pybamm.Interpolant(time_profile[:],current_profile[:],
pybamm.t)

# Set drive cycle
parameter_values=pybamm.ParameterValues("Chen2020")
parameter_values["Current function [A]"] = current_interpolant

model = pybamm.lithium_ion.SPMe()
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve()
sim.plot(["Current [A]", "Battery voltage [V]"])

tinosulzer, 2023-06-10:
Your MWE works fine for me, so I think you just need to update your version of pybamm

closed 2023-08-08 by github-actions[bot]

chetandharmadhikari, 2023-08-08:
Hi,

I am able to resolve the problem after updating the software. Please close
the issue.

Thank Ms for your help,

Chetan