pybamm-team/PyBaMM #3035: Splitting the required dependencies into optional dependencies

arjxn-py, 2023-06-14:
### Description

Currently pybamm is installing many dependencies we want to have a lite version of pybamm by splitting the required deps into optional deps

Saransh-cpp, 2023-06-14:
Related to #2457

Saransh-cpp, 2023-06-14:
Discussion from today's meeting -
- Making dependencies like `imageio` optional like `jupyter`
  - Adding multiple small optional tags like `[gif]` for `imageio`
  - Adding a single big tag to cover all the additional dependencies, like `[all_optional]`
  - Visualise dependencies using https://pypi.org/project/pipdeptree/ 

Maintainers can iteratively add here the dependencies which they think can be made optional.

tinosulzer, 2023-06-14:
The following dependencies can be made optional without affecting 95% of functionality:
- pandas
- anytree
- autograd
- scikit-fem
- pybtex
- sympy

agriyakhetarpal, 2023-06-14:
The dependency graph with `pipdeptree` is also related to the documentation project, I may pick that up later if that's okay?

arjxn-py, 2023-06-14:
> The following dependencies can be made optional without affecting 95% of functionality

How should I go with grouping them further or should we have a single tag say `dev`?

brosaplanella, 2023-06-14:
> The following dependencies can be made optional without affecting 95% of functionality:
> 
> * pandas

About pandas, I think the trickiest point will be that some parameter sets won't work without pandas (they use `read_csv`). One way around would be to hardcode the csv files in the Python script for the given parameter.

Just to clarify, I am not against pandas being optional, but I think it would be good to keep all parameters working with the basic PyBaMM.

tinosulzer, 2023-06-14:
For pandas we could set the default to use the equivalent numpy function and then we would not need pandas

I would actually put tqdm and bpx in the easy bucket too

For the hard ones:
- scipy: impossible to get rid of if solving PDEs (it is used in the discretization)
- casadi: very hard to get rid of and would lead to performance regressions
- xarray: impossible to get rid of without having annoying-to-maintain try-excepts in the `ProcessedVariable` class

Saransh-cpp, 2023-06-14:
> How should I go with grouping them further or should we have a single tag say dev?

There should be smaller groups, like `[plot]` for `matplotlib` and `imageio` and `[jax]` for `jax` and `jaxlib`, and a larger group, like `[optionals]` with every optional library included (maybe not the ones that are OS specific).

tinosulzer, 2023-06-14:
I really like the way pandas handles and documents optional dependencies: https://pandas.pydata.org/docs/getting_started/install.html#optional-dependencies

So `pybamm[all]` would install everything.

I'm not sure whether zsh handles square brackets so we might need to encourage using quotes e.g. `pip install "pybamm[all]"` (as pandas does)

arjxn-py, 2023-06-16:
Pandas handles this type of installation using [`pyproject.toml`](https://pip.pypa.io/en/stable/reference/build-system/pyproject-toml/) introduced in [PEP 518](https://peps.python.org/pep-0518/) but our build-system is based on `setup.py` which I'd also need to figure out to migrate to `pyproject.toml`

Saransh-cpp, 2023-06-16:
You don't necessarily need `pyproject.toml` to create optional dependencies. Migrating to `pyproject.toml` would be a bit tricky (we've tried once 🙂), but yes, very much appreciated. I'm not sure which build backend is used by `pandas` but you can look into `setuptools`, `scikit-build`, and `meson-py`.

PR #3044 (2023-07-06): Create Optional Dependencies
# Description

The goal of this PR is to make most of the dependencies optional & split them in specific group

Fixes #3035 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [x] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an [... cut]
closed 2023-07-06 by Saransh-cpp