pybamm-team/PyBaMM #3036: Print names for equivalent circuit models

rtimms, 2023-06-14:
Most of the variables haven't been assigned print names for the equivalent circuit models, so the latex output just prints long names instead of symbols, see
```python3
import pybamm

model = pybamm.equivalent_circuit.Thevenin()
model.latexify()
```

kumaryash7, 2023-09-04:
@rtimms Hello, myself Yash from India. I am new to the open source. Can you assign me this issue? I have already set up the project.

medha-14, 2024-12-19:
I assigned `print_names` to the variables directly in the `update` method, and it worked. However, I believe there might be an existing method or mechanism that ensures consistent naming of variables, especially for generating LaTeX representations, which I may have overlooked.Can someone please clarify this ?