pybamm-team/PyBaMM #3081: Improve memory usage for experiments with `start_time`

brosaplanella, 2023-06-29:
At the moment, two steps with the same operating conditions but different `start_time` are treated as different. This means that the model needs to be processed twice, which can end up with a big memory consumption (and longer running times) for very long experiments.

A way to improve this would be to read the `start_time` from the unprocessed step, this way we can have a single processed step for all of them.

brosaplanella, 2023-06-29:
A key point here is how we define the `__eq__` method for the `_Step` class, as that's what is used to determine unique steps. The original method that @tinosulzer wrote basically checked if both objects are a `_Step` and have the same `__repr__`. I added the condition that `next_start_time` and `end_time` should be equal too.

I think we should move to checking specific variables in the class to determine if two steps are identical (rather than the `__repr__`), so just wanted some input on which ones we should check. I am thinking of

- `typ`
- `value`
- `duration`
- `termination`
- `period`
- `temperature`

This would leave out (from those displayed in `__repr__`):

- `tags`
- `start_time`
- `description`

My cut has been fairly conservative, as probably some of those in the first category could also be left out.

tinosulzer, 2023-06-29:
To decide if two steps should be "equal" we have to look at the experiment builder in the `Simulation` class. Different steps can either be processed differently, or they can be processed the same and differentiated by inputs passed in at solve time. 
Inputs that affect how the step is processed are
- typ
- value
- termination
- temperature

Inputs that get passed in at solve time are
- duration
- period
- start time

Inputs that are just descriptive (not explicitly used for solving) are
- tags
- description

PR #3261 (2023-09-19): Improve memory use in experiments
# Description

Improve memory use in experiments by redefining what steps are unique. Two steps are considered the same if `type`, `value`, `termination` and `temperature` are the same, as these are the variables used when processing the model. The remaining variables are only used when running the model and can be passed as inputs. I have also defined a `basic_repr` which returns a representation with only those variables, while `__repr__` returns a representation with all the variables.

Fixes #3081 

## Type of change

Please add a line in the relevant sect [... cut]
closed 2023-09-19 by brosaplanella