pybamm-team/PyBaMM #3100: [Bug]: IDAKLU solver error

Mrzhang-hub, 2023-07-04:
### PyBaMM Version

23.4.1

### Python Version

3.9

### Describe the bug

RuntimeError:C:\..\casadi\core\serialization_stream.cpp:355:Assertion"load_version==v" failed:ProtoFunction failed. Object written in version 2 but can only read in version 1.

### Steps to Reproduce

```
import pybamm
import numpy as np
model = pybamm.lithium_ion.SPM()
geometry = model.default_geometry
param = model.default_parameter_values
param.process_model(model)
param.process_geometry(geometry)
mesh = pybamm.Mesh(geometry,model.default_submesh_types,model.default_var_pts)
disc = pybamm.Discretisation(mesh,model.default_spatial_methods)
disc.process_model(model)
t_eval = np.linspace(0,3600,100)
if pybamm.have_idaklu():
     klu_sol = pybamm.IDAKLUSolver(atol=1e-8,rtol=1e-8).solve(model,t_eval)
plot = pybamm.Quickplot(klu_sol)
plot.dynamic_plot()
```

brosaplanella, 2023-07-04:
Strange, it works at my end (even with version 23.4.1). Could be that when you upgraded to the newer PyBaMM version, casadi didn't. Can you try upgrading both PyBaMM and Casadi to their respective latest versions?

Mrzhang-hub, 2023-07-05:
> 

I have upgrade pybamm and casadi to their newest version.(pybamm=23.5 casadi=3.6.3) But the problem still exist. Is that has something to do with the version of c++?

Mrzhang-hub, 2023-07-05:
I downgraded  pybamm to version 22.10 and casadi to version 3.5.5, which solved the problem.

tinosulzer, 2023-07-27:
Reopening as suggested in https://github.com/pybamm-team/PyBaMM/discussions/3193#discussioncomment-6563103

We need to find out why the newer pybamm or casadi version breaks this

jsbrittain, 2023-12-01:
Reopening as the issue has reemerged, including in pip installs of pybamm. See #3193 for related discussion.

jsbrittain, 2023-12-01:
Some progress - Given the new information I have managed to recreate the serialisation error on my (working) machine by taking the following actions:
1. Removed/renamed `~/local` to remove any existing (local) `casadi` installs.
2. Build `casadi` from source following [their instructions](https://github.com/casadi/casadi/wiki/SourceBuild) (this required casadi version 3.5.5 which builds `libcasadi.3.6.dylib`, not version 3.6.0 which actually builds `libcasadi.3.7.dylib`.
3. Reinstall PyBaMM using nox (`pybamm-requires` then `dev`)
4. Test the installation using the script provided in #3193 

After that I receive the `Assertion "load_version==v" failed:
DeSerialization of ProtoFunction failed. Object written in version 2 but can only read in version 1.` error. This only impacts the new build/install, my previous installation still works fine.

So it seems that this error is in-fact due to a mismatch in the version of `casadi` used in Python conflicting with the version that IDAKLU uses to link at build time. Removing `casadi` installs outside of PyBaMM prior to install should resolve this, but I acknowledge that this is not an ideal solution and the installer should instead prioritise the python install over the system install to maintain synchrony. Further, it is not clear to me how or why this is impacting the wheels (did the wheel itself build using two `casadi` versions?). Now that we have a reprodicuble testbed for examining the issue we should be able to make further progress.

pipliggins, 2023-12-01:
@jsbrittain @agriyakhetarpal The `CMakeLists.txt` fix works for me too!

jsbrittain, 2023-12-07:
@agriyakhetarpal Removing the registry definitely sounds like the most maintainable solution going forwards. I took a quick look at the error from your Windows wheel build with python-casadi and, comparing it to the PyPI package listings, it looks as though there may be a misreferenced file in the casadi cmake targets (`libcasadi.lib` is requested, but only `casadi.lib` exists); I'll need to get hold of a Windows machine to see if this actually resolves the build issue, but even if it does that would mean an upstream change in the casadi codebase (we might be able to hotfix it at our end in the meantime, but that wouldn't be stable for subsequent releases). Have you tried this with the latest casadi release (3.6.4)?

agriyakhetarpal, 2023-12-07:
I haven't tried it with the latest `casadi` release yet since they don't seem to provide any minor version compatibility of sorts (based on the many, many changes between `3.6.3` and `3.6.4` as noted in the release notes); but there should be no harm in trying that—I will update my fork of the registry again, try both the Python and vcpkg casadi and get back to this—maybe they fixed this on their end. I had chosen `3.6.3` specifically to synchronise the versions for the build-time and run-time dependencies, but I suppose `pip` downloads the latest version anyway unless we don't pin them.

chmabaur, 2023-12-08:
@agriyakhetarpal Thanks a lot for your efforts to solve this issue! I have tested both wheel files in separate virtual environments on Windows for python version 3.9. I am glad to say that in both cases the serialization error disappeared.

I am not able to follow your discussions above, but I hope it is ok that `pip show` prints the `casadi version 3.6.4` in both cases.

agriyakhetarpal, 2023-12-08:
Thanks for confirming! Yes, `pip show casadi` displaying `3.6.4` as the output is fine. I guess the gist is that the `casadi` has to be `3.6.X` for the serialisation to work properly and a difference between the patch versions at runtime should be okay, but linking and running with different minor versions (say, `3.5.X` vs `3.6`) might cause issues as has been the case here.

Tagging @martinjrobins to request for write access to the casadi vcpkg registry (I think merging a PR from my end there will require a further, redundant update to the git-trees so it would be great to directly push) and then this issue can be closed with an update to the baseline commit in the vcpkg configuration file.

P.S. I will open a new, separate issue to explore the possibility of moving away from the registry and getting back to using the Python `casadi` again. It looks like it can be fixed with a patch on our end, but I don't have enough experience to comment about that procedure haha

PR #3637 (2024-01-06): Update to `casadi` `3.6.4` for Windows wheels
# Description

See #3193 and #3100. The `pybamm-team/casadi-vcpkg-registry` has been updated to use `casadi==3.6.4` for linkage in the Windows wheels. We should ensure that the version of `casadi` used for IDAKLU compilation remains the same across platforms and architectures, for which #3603 has been opened.

The wheels have been built with this update earlier on my fork and had passed as noted in [#3100](https://github.com/pybamm-team/PyBaMM/issues/3100/#issuecomment-1846886211). The issue has been noted to be fixed for the Windows wheels with th [... cut]
closed 2024-01-06 by Saransh-cpp