pybamm-team/PyBaMM #3101: [Bug]: `starting_solution` does not work if last cycle is not saved

brosaplanella, 2023-07-04:
### PyBaMM Version

23.5

### Python Version

3.9.17

### Describe the bug

When using `starting_solution` in solving a simulation, if the last cycle of the `starting_solution` is not saved, then the initialisation doesn't work: the new solution starts at `t=0` and does not contain the previous states (not even the first cycle which should be always saved).

My guess is that when checking the solution for the last cycle it finds it is None and thus it treats it as if no `starting_solution` has been provided.

### Steps to Reproduce

```python3
        model = pybamm.lithium_ion.SPM()

        experiment = pybamm.Experiment([
            # pybamm.step.string("Discharge at C/2 for 10 minutes", start_time=datetime(2023, 1, 1, 8, 0, 0)),
            pybamm.step.string("Discharge at C/2 for 10 minutes"),
            pybamm.step.string("Rest for 10 minutes"),
        ])

        sim = pybamm.Simulation(model, experiment=experiment)
        solution = sim.solve(save_at_cycles=[1])

        experiment = pybamm.Experiment([
            # pybamm.step.string("Discharge at C/2 for 20 minutes", start_time=datetime(2023, 1, 1, 10, 0, 0)),
            pybamm.step.string("Discharge at C/2 for 20 minutes"),
            pybamm.step.string("Rest for 20 minutes"),
        ])

        sim = pybamm.Simulation(model, experiment=experiment)
        new_solution = sim.solve(calc_esoh=False, starting_solution=solution)
```

Plotting `new_solution` yields:
[image]

If, instead, I do `save_at_cycles=None` in the first solve, I get the expected result:
[image]

brosaplanella, 2023-07-04:
Maybe not strictly related to this, but another issue is that the way `starting_time` is currently implemented means that the `initial_start_time` is taken from the experiment, but if an initial solution is provided it should be taken from there.

tinosulzer, 2023-07-04:
It should be reasonably easy to fix, we just need to save `last_state` regardless of whether the last cycle is saved and then use that for the starting solution

brosaplanella, 2023-07-07:
This issue might also align with #2484

PR #3177 (2023-09-15): Fix `starting_solution` issues
# Description

This PR is to solve the issue with the `starting_solution` when the final cycle was not saved. As discussed in #3101, I think the most robust way forward is to always save the final cycle (even at the expense of consuming a bit more of memory), as it also fixes the `last_state`.

It also fixes another bug (described in the same issue) which caused the simulation to always take the `initial_start_time` from the experiment, but it should be taken from the `starting_solution` if it exists.

Fixes #3101

## Type of change

Please add a line in the rele [... cut]
closed 2023-09-15 by rtimms