pybamm-team/PyBaMM #3114: [Bug]: Coordinate systems of variables do not get checked against known

brosaplanella, 2023-07-06:
### PyBaMM Version

23.5

### Python Version

3.9.16

### Describe the bug

Currently the coordinate system of a variable is not checked against the known system. This means that if the wrong argument is passed (e.g. due to a typo) it will default to `"cartesian"` without throwing an error.

### Steps to Reproduce

The MWE would be very lengthy, but basically it can be checked that:

1. The only place where `KNOWN_COORD_SYS` is checked is to test meshes for the battery model (i.e. it does not actually check for wrong coordinate systems).
2. In `finite_volume.py` there is an else statement that means that anything not being cylindrical polar or spherical polar will default to cartesian.

brosaplanella, 2023-07-24:
Ooops, that was already reported in #2220 (contains some additional info). Let's make sure we close both issues when we merge PR.

RuiheLi, 2023-07-25:
Yes, I will. Will look into #2220. Thanks for the remind! @brosaplanella

RuiheLi, 2023-07-25:
> Yes, I will. Will look into #2220. Thanks for the remind! @brosaplanella

Just checked the time of both issue, will look into both.

rtimms, 2024-11-22:
closing, see #4600

closed 2024-11-22 by rtimms