pybamm-team/PyBaMM #3124: [Bug]: benchmarks are failing

agriyakhetarpal, 2023-07-08:
### PyBaMM Version

23.5

### Python Version

all

### Describe the bug

The [scheduled benchmarks](https://github.com/pybamm-team/PyBaMM/actions/runs/5492426530) and the benchmarks on PRs are failing because the `asv` configuration fails to find `pybtex`, since it is not a core dependency but has been shifted to the `[cite]` extras group after #3044 was merged

### Steps to Reproduce

Via the `benchmarks_on_push.yml` and `periodic_benchmarks.yml` workflow files

agriyakhetarpal, 2023-07-08:
I tried adding `pybtex` to the Python dependencies in https://github.com/pybamm-team/PyBaMM/commit/59fa13d4411b5973767eb8a3c56e9066ce768ad4, but that did not work—is there a way to install the extras from a built wheel file?

Saransh-cpp, 2023-07-08:
You'll probably need to add `pybtex` in https://github.com/pybamm-team/PyBaMM/blob/develop/asv.conf.json

PR #3145 (2023-07-18): Fix Failing Benchmarks
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #3124 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change t [... cut]
closed 2023-07-18 by Saransh-cpp