pybamm-team/PyBaMM #3130: [Bug]: 'Multiplication' object has no attribute 'y_slices'

Mrzhang-hub, 2023-07-10:
### PyBaMM Version

23.5

### Python Version

3.9

### Describe the bug

```
  File "D:\ProgramData\Anaconda3\lib\site-packages\pybamm\solvers\solution.py", line 108, in __init__
    self.check_ys_are_not_too_large()
  File "D:\ProgramData\Anaconda3\lib\site-packages\pybamm\solvers\solution.py", line 311, in check_ys_are_not_too_large
    y_var = y[model.variables[var.name].y_slices[0]]
AttributeError: 'Multiplication' object has no attribute 'y_slices'
```

### Steps to Reproduce

```
import numpy as np
import os
import pybamm
import pandas as pd
import matplotlib.pyplot as plt
model = pybamm.lithium_ion.SPMe(
    options={
        "thermal": "lumped",
        "dimensionality": 0,
        "cell geometry": "arbitrary",
        "electrolyte conductivity": "integrated",
    },
    name="TSPMe",
)
param = pybamm.ParameterValues("Chen2020")
param.update({'Current function [A]':'[input]'})
param.update({"Negative electrode diffusivity [m2.s-1]": '[input]'})
param.update({"Positive electrode diffusivity [m2.s-1]": '[input]'})
geometry = model.default_geometry
param.process_geometry(geometry)
param.process_model(model)
var = pybamm.standard_spatial_vars
var_pts = {var.x_n:30,var.x_s:30,var.x_p:30,var.r_n:10,var.r_p:10}
mesh = pybamm.Mesh(geometry,model.default_submesh_types,var_pts)
disc = pybamm.Discretisation(mesh,model.default_spatial_methods)
disc.process_model(model)
t_eval = np.linspace(0,3600,100)
solver = pybamm.CasadiSolver(mode='safe',atol=1e-6,rtol=1e-3)
start_params = [0.15652, 3.54e-14, 1.01e-13]
solution = solver.solve(model,t_eval,inputs={'Current function [A]': start_params[0],
                "Negative electrode diffusivity [m2.s-1]": start_params[1],
                "Positive electrode diffusivity [m2.s-1]": start_params[2]},
                calculate_sensitivities=True)
fig,ax = plt.subplots(1,3,figsize=(10,5))
ax[0].plot(t_eval,solution['Current [A]'].entries)
ax[0].set_xlabel(r'$t$')
ax[0].set_ylabel(r'$Current [A]$')
ax[1].plot(t_eval,solution['Terminal voltage [V]'].entries)
ax[1].set_xlabel(r'$t$')
ax[1].set_ylabel(r'$Terminal voltage [V]$')
ax[2].plot(t_eval,solution['Terminal voltage [V]'].sensitivities['Current function [A]'])
ax[2].set_xlabel(r'$t$')
ax[2].set_ylabel(r'$sensitivities of Terminal voltage wrt Current function$')
plt.tight_layout()
plt.show()
```

brosaplanella, 2023-07-17:
I have now updated the MWE, it seems to be an issue with the calculation of the sensitivities. @martinjrobins is this a bug?

brosaplanella, 2023-07-27:
I have now noticed that this has to do with the variables being too large. Can you set `calculate_sensitivities=False` and see if the issue persists (so we know if it is with solving the model or with the sensitivities only).

Mrzhang-hub, 2023-07-27:
1. Yes, I have also noticed that  `y > pybamm.settings.max_y_value` when `check_ys_are_not_too_large`, but I do not know why it is that large and why this error `AttributeError: 'Multiplication' object has no attribute 'y_slices'` appears. 
2. When I set `calculate_sensitivities=False`, it runs normally. Thus it is caused by sensitivities only.  The following figure is the simulation result.
[image]

brosaplanella, 2023-07-28:
Thanks for the update. I will change the title to something more informative.

martinjrobins, 2023-09-07:
not sure why this code was trying to get the `y_slice` from the variable, but I've changed so that it is getting it from `model.y_slices`, which fixes the issue, see pr #3313

martinjrobins, 2023-09-07:
> not sure why this code was trying to get the `y_slice` from the variable, but I've changed so that it is getting it from `model.y_slices`, which fixes the issue, see pr #3313

Ahhh, this was for models that don't need to be discretised. I've also done a fix for this

PR #3313 (2023-09-08): I3130-y-slice-bug
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #3130

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ pre-co [... cut]
closed 2023-09-08 by martinjrobins