pybamm-team/PyBaMM #3131: Assign a spacial temperature vector as initial temperatrue field for DFN model

xyyrly, 2023-07-10:
### Description

Hi team,
I try to set an initial temperature distribution in x dimension for DFN but now it looks only a scalar value can be set for the whole domain. Could it be fixed?   Thanks!

tommaull, 2023-07-11:
I'll take this one

xyyrly, 2023-07-12:
Thank you very much!

tommaull, 2023-07-12:
xyyrly - assuming you were using the xfull temperature model?

xyyrly, 2023-07-12:
Hi tommaull,

I am using a new thermal model to calculate electrolyte temperature along x dimension and particle temperature along r dimension,  kind of like DFN model.

tommaull, 2023-07-12:
Can you write out the steps to replicate with existing models?

xyyrly, 2023-07-12:
Oh sorry I was confused. I think here what I hope to realize is to assign a spatical temperature distribution to DFN model without subthermal model. Is it possible?

tommaull, 2023-07-12:
I think a thermal model is required. My approach was going to be to add something like: 

def set_initial_conditions(self, variables):
        T = variables["Cell temperature [K]"]

        if isinstance(self.param.T, (list, np.ndarray)):
            if len(self.param.T) == 3:
                # Get sizes of negative, separator, and positive domains
                sizes = {
                    "negative electrode": self.mesh["negative electrode"].npts,
                    "separator": self.mesh["separator"].npts,
                    "positive electrode": self.mesh["positive electrode"].npts,
                }

                # Check if sizes match the length of T_init
                if all(size == len(T_init) for size in sizes.values()):
                    self.initial_conditions = {
                        "negative electrode": self.param.T_init[
                            : sizes["negative electrode"]
                        ],
                        "separator": self.param.T_init[
                            sizes["negative electrode"] : sizes["negative electrode"]
                            + sizes["separator"]
                        ],
                        "positive electrode": self.param.T_init[
                            sizes["negative electrode"] + sizes["separator"] :
                        ],
                    }
                else:
                    raise ValueError("T_init vector size does not match domain sizes")
            else:
                raise ValueError("T_init should be a vector of length 3")
        else:
            self.initial_conditions = {T: np.full_like(T, self.param.T_init)}

rtimms, 2023-07-12:
@tommaull take a look at e.g. how we define [initial concentration](https://github.com/pybamm-team/PyBaMM/blob/7fe7fde01d79d9b9857663660e014db7c8e0fbaa/pybamm/parameters/lithium_ion_parameters.py#L495), which can depend on `x` and `r`, and then how it is [used in the model](https://github.com/pybamm-team/PyBaMM/blob/7fe7fde01d79d9b9857663660e014db7c8e0fbaa/pybamm/models/submodels/particle/fickian_diffusion.py#L297).

You'll need to make [initial temperature](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/thermal_parameters.py#L42) a `FunctionParameter`, then you can pass `T(x)` as an initial condition.

tommaull, 2023-07-16:
Put in a draft PR to show progress but having an issue I don't understand in the expression tree.