pybamm-team/PyBaMM #3133: create calculate_sensitivities for experiment

Mrzhang-hub, 2023-07-10:
### Description

It seems now that pybamm does not support calculate_sensitivities method for experiment. However, when working on parameter estimation with gradient-based method, it is needed to access parameter sensitivities(partial derivative of terminal voltage to parameters) under certain charging/discharging experiment. So is it possible to add calculate_sensitivities for experiment? @brosaplanella  @tinosulzer

tinosulzer, 2023-07-10:
I don't know how we would do this because we'd have to differentiate through the events (such as the voltage cut-offs), which as far as I know is not possible

martinjrobins, 2023-07-10:
agree, I think in this case you would have to do the parameter estimation over each separate part of the experiment (i.e. between the events)

closed 2023-07-10 by tinosulzer