pybamm-team/PyBaMM #3140: [Bug]: Integration tests failing M-Series Hardware

BradyPlanden, 2023-07-11:
### PyBaMM Version

23.5

### Python Version

3.8-3.11

### Describe the bug

Integration tests are currently failing on apple m-series hardware. This seems to be centred around reaching the maximum number of timestep decreases for the casadi solver when testing the cracking submodels. A quick fix might be to decrease the intial step size when running [these](https://github.com/pybamm-team/PyBaMM/blob/1a012e75953d4cd83e6705b4e5c250291fbe75fd/tests/integration/test_models/test_full_battery_models/test_lithium_ion/base_lithium_ion_tests.py#L237C3-L237C3) tests.

### Steps to Reproduce

Run `nox -s integration` on m-series hardware.

### Relevant log output

```shell
https://github.com/pybamm-team/PyBaMM/actions/runs/5515391562/jobs/10061099876
```

brosaplanella, 2023-07-24:
Sounds like a good fix! Can you check if it works locally? Note that you can pass kwargs to the simulation through `run_basic_processing_test`.

BradyPlanden, 2023-07-26:
I can take a look at this later in the week. Thinking on it further, we probably need to take a look at why the casadi is decreasing the timesteps to the point of an error for Apple ARM hardware, but not on x64. I'll probe a bit further and come back to this issue.

kratman, 2023-09-27:
I was looking at this for the examples (```nox -s examples```) which fail on my M1 Mac. I had no luck solving this by changing the dt_max parameter.

I started with **SEI-on-cracks.ipynb** and got it working. I am not sure why yet, but I found that I could stop this error by changing:
```python
var_pts = {
    "x_n": 20,  # negative electrode
    "x_s": 20,  # separator 
    "x_p": 20,  # positive electrode
    "r_n": 30,  # negative particle
    "r_p": 30,  # positive particle
}
```
to
```python
var_pts = {
    "x_n": 20,  # negative electrode
    "x_s": 20,  # separator 
    "x_p": 20,  # positive electrode
    "r_n": 26,  # negative particle
    "r_p": 26,  # positive particle
}
```
Reducing the size of the mesh seems to make it work. Not the best solution, but it lets the tests finish for now.
Adding the same var_pts to **submodel_cracking_DFN_or_SPM.ipynb** also makes that one start to work. Finally I did the same thing in the integration tests, and the fix works there as well.

I made a PR so the stop-gap solution could be put into place. I did not mark the PR as fixing this issue since it only gets the tests running.

closed 2024-06-05 by kratman