pybamm-team/PyBaMM #3164: Improvements to plotting functionality

brosaplanella, 2023-07-19:
### Description

Here are a list of possible improvements to PyBaMM's plotting capability (please suggest additional ones if you have any ideas):
* Allow `QuickPlot.plot` to take a list of times to plot various snapshots.
* Have a `static_plot` mode so variables that depend on space (1D) and time are plotted as heatmaps rather than dynamic curves.

### Additional context

Not sure if there are easy hacks or ways around this, but if so it was not clear in the docs.

Saransh-cpp, 2023-07-20:
I think a portion of `create_gif` does acts as the proposed `QuickPlot.plot`. We can maybe refactor this method to make way for the new method -

https://github.com/pybamm-team/PyBaMM/blob/e730dcb7cfc7c6da595618502d4e026378c0d4f4/pybamm/plotting/quick_plot.py#L745-L772

brosaplanella, 2023-07-20:
My bad, I wasn't very clear. I meant multiple snapshots on the same plot.

medha-14, 2024-11-22:
I would also like to work on the second issue, which involves creating a `static_plot` function. Could you please provide a more detailed explanation of this issue to help me better understand the requirements?