pybamm-team/PyBaMM #3165: Some notebook lines are not being tested

brosaplanella, 2023-07-19:
As spotted in https://github.com/pybamm-team/PyBaMM/pull/3110#discussion_r1267382517, currently some lines in the notebooks are not tested. I don't know the reason, but I suspect that it is because in [this example](https://github.com/pybamm-team/PyBaMM/blob/v23.5/docs/source/examples/notebooks/creating_models/6-a-simple-SEI-model.ipynb) the following lines are not being tested
```python3
import ipywidgets as widgets
widgets.interact(plot, t=widgets.FloatSlider(min=0,max=solution.t[-1]*1e6,step=0.1,value=0));
```
so the error is not detected.

I am not sure if we were aware of this or not. I will work with @arjxn-py to fix that particular example, but this issue is to discuss if we need to fix the notebook testing.

agriyakhetarpal, 2023-07-21:
There is also another issue with notebooks that use `ipywidgets` in specific (which was discussed with @tinosulzer and @rtimms in our last meeting)—apparently `nbsphinx` omits the slider on those notebooks when converting them to `.rst` format for rendering on the documentation, but it uses Pandoc to do it, so it could be out of our control.

I would propose we should use `importlib` and check for the use of `ipywidgets` (i.e., these lines) in any of the code blocks (which should be easy to add since we use `nbconvert` to parse the code portions and convert them to `.py` files) and check if the output of the slider is correct (I don't know how to do this)

brosaplanella, 2023-07-26:
An option would be to use [nbmake](https://github.com/treebeardtech/nbmake) to test the notebooks, instead of what we currently do.

arjxn-py, 2023-07-26:
> An option would be to use [nbmake](https://github.com/treebeardtech/nbmake) to test the notebooks, instead of what we currently do.

[nbmake](https://github.com/treebeardtech/nbmake) looks promising.
I had also went through one - [nbval](https://github.com/computationalmodelling/nbval)

PR #3344 (2023-09-22): Revamped notebook testing infrastructure
# Description

Adds `pytest` plus the `pytest-xdist` and `nbmake` plugins to execute notebooks in parallel, along with associated and ancillary changes to the documentation and infrastructure

Fixes #3165 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [x] Optimizatio [... cut]
closed 2023-09-22 by Saransh-cpp