pybamm-team/PyBaMM #3187: Implement GITT model

brosaplanella, 2023-07-24:
### Description

For the GITT simple model, we need to implement a linearised voltage

$$ V = U (c_0) + U'(c_0) (c - c_0) + R I $$

(note that there are some subtleties on what goes where...).

### Motivation

Estimating the diffusion coefficient from GITT data can be improved by using SPM with linearised voltage.

### Possible Implementation

We need a linearised OCP and a linearised inverse Butler-Volmer (some might already exist).

RuiheLi, 2023-09-29:
Hi @brosaplanella I am looking into this now. May I ask the linearised voltage V you mentioned refer to the "Terminal voltage [V]", and the main reason that is can be written in that way is that we linearise the input OCP and inverse Butler-Volmer. If so, what I need to do is to create a linearised OCP function (for example, base on `graphite_LGM50_ocp_Chen2020(sto)`), then for the inverse Butler-Volmer, create a new class in pybamm/models/submodels/interface/kinetics/inverse_kinetics/inverse_butler_volmer.py like `class linearised_InverseButlerVolmer(BaseInterface):`?

brosaplanella, 2023-10-02:
PyBaMM already has a `Linear` kinetics class, which is linearised Butler-Volmer, so it would be a matter of implementing the linearised voltage. The issue is that the voltage is pulled from the electrode potentials, so this would require a linearised electrode potential model (as in, linearised OCV).

RuiheLi, 2023-10-08:
> PyBaMM already has a `Linear` kinetics class, which is linearised Butler-Volmer, so it would be a matter of implementing the linearised voltage. The issue is that the voltage is pulled from the [...]

@brosaplanella  Do you think it makes sense to use a global expression for the linearised OCV (the one you proposed)? I reckon that it might be too simple to reproduce / fit decently the complext shape of the OCP curves for either electrodes. Would you recommend we use piece-wise linearised OCV? If so, then the problems become how many pieces we should pick...

brosaplanella, 2023-10-09:
I don't expect this model to be used for anything else other than GITT simulations & data fitting, but you are right and it might be quite confusing for users. If you want let's leave it for now, feel free to choose another issue if you want.