pybamm-team/PyBaMM #3195: Clean up GNU-Linux and macOS installation page

Saransh-cpp, 2023-07-27:
Right now, the Linux and macOS installation page has a "[Developer Install](https://docs.pybamm.org/en/latest/source/user_guide/installation/GNU-linux.html#developer-install)" section, which should not be there. The contents should either be moved to the "[Install from source](https://docs.pybamm.org/en/latest/source/user_guide/installation/install-from-source.html)" guide, or removed if they are already present there.

@arjxn-py this should be a quick fix, feel free to pick this up 🙂

Saransh-cpp, 2023-07-27:
Also -

- "Install from source (developer install)" should be renamed to something like "Install from source (GNU Linux and macOS)"
- "Windows Subsystem for Linux (WSL)" should be renamed to "Install from source (Windows)"

The text should be cleaned up as well. For instance https://docs.pybamm.org/en/latest/source/user_guide/installation/windows.html#user-install mentions -

> PyBaMM’s dependencies (such as numpy, scipy, etc) will be installed automatically when you install PyBaMM using pip

which is not the case with `scipy` now. The Windows section should also be extended to depict user installation in WSL, and not the developer installation.

Similarly, the user installation "troubleshooting" lists -

> Problem: I’ve made edits to source files in PyBaMM, but these are not being used when I run my Python script.
>
> Solution: Make sure you have installed PyBaMM using the -e flag, i.e. pip install -e .. This sets the installed location of the source files to your current directory.

which should be in the developer installation section.

In a nutshell, the developer and user installation docs are tangled with each other right now, which is not at all good.

Saransh-cpp, 2023-07-27:
> Similarly, the user installation "troubleshooting" lists -
>
> which should be in the developer installation section.

This might also be the reason why users add their own code to PyBaMM (locally, with no intentions to push) instead of using it as a library, as mentioned by @tinosulzer.

arjxn-py, 2023-08-04:
> > PyBaMM’s dependencies (such as numpy, scipy, etc) will be installed automatically when you install PyBaMM using pip
> 
> which is not the case with `scipy` now.
Scipy is still a required dependency so this should be good for now.

arjxn-py, 2023-08-04:
>  The contents should either be moved to the "[Install from source](https://docs.pybamm.org/en/latest/source/user_guide/installation/install-from-source.html)" guide, or removed if they are already present there.

Since [Install from source](https://docs.pybamm.org/en/latest/source/user_guide/installation/install-from-source.html) already covers that section so it was redundant.

PR #3235 (2023-08-29): Clean up installation guides
# Description

Since the developer and user installation docs are tangled with each other right now, this PR aims to simplify them.

Fixes #3195 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking  [... cut]