pybamm-team/PyBaMM #3196: pybamm.Simulation - private attributes ignore current state of public attributes in build() and solve()

ejfdickinson, 2023-07-27:
In the `pybamm.Simulation` class, some behaviour of implicitly public attributes is ambiguous / poorly documented.

In particular, the `build()` and `solve()` functions assume that the values of `self.parameter_values` and `self.model` equal those in the initial state of the `pybamm.Simulation` instance following `__init__()`, since they act on private attributes that are set to the initial state of theser variables. This conflicts with the API's implication that these attributes are user-editable.

Example:

```
model = pybamm.lithium_ion.DFN()
# User wants to discard automatically generated events
model.events = []
sim = pybamm.Simulation(
   model,
   pybamm.ParameterValues("Chen2020")
)
sim.solve()    # Runs with no events
```

whereas

```
model = pybamm.lithium_ion.DFN()
sim = pybamm.Simulation(
   model,
   pybamm.ParameterValues("Chen2020")
)
# User wants to discard automatically generated events
sim.model.events = []
sim.solve()     # Runs with events anyway!
```

This is because `solve()` replaces `self.model` using a call to `set_parameters()` with `sim._unprocessed_model`, which preserves the state of `model` when `sim` was initialised.

Other affected attributes:

* `self._unprocessed_parameter_values` - preserves initial state of `self.parameter_values`, irrespective of any subsequent changes
* `self.operating_mode` - determined based on the initial condition, irrespective of any subsequent changes

Better API behaviour:

* Re-initialise private attributes based on the public state of the at time of use, not at time of initialisation.
* Or, if direct edits to `sim.parameter_values` and `sim.model` are strictly not allowed after `sim` is initialised, they should be declared private (e.g. `sim._model`).

In any case, the API documentation could be clearer about the activities of `set_parameters()` and `build()`, and when direct external calls are required to these functions.

tinosulzer, 2023-07-28:
Sounds reasonable, thanks for the suggestion

rtimms, 2023-07-28:
My vote is for removing all the setter methods. Not sure we want to allow any of these properties to be edited after the simulation has been created.

tinosulzer, 2023-07-28:
Agreed. We could even make it impossible by overloading `__setattr__`, but I was reading up on this and the "pythonic" way is to allow it but make it obvious it is "at your own risk" if you use an attribute that starts with an underscore

tinosulzer, 2023-07-30:
@agriyakhetarpal this aligns with what we've been discussing about private vs public API, can you take a look at this?

brosaplanella, 2023-08-03:
The only part I could think of changing are the parameters, to set some to inputs (we do this in pbparam). I opened an issue about this (https://github.com/pybamm-team/PyBaMM/issues/3113). But I agree that any changes to a simulation should be "at your own risk".

PR #3267 (2023-09-21): Disallow modifications to `pybamm.Simulation` object attributes after initialisation
# Description

Removes the setter methods in the `Simulation` class, so that parameters defined in the initialisation of a `pybamm.Simulation` instance cannot be modified unless the instance is redefined

This way, attributes like `sim.model` or `sim.parameter_values` cannot be edited directly. A new `sim` object with the modified parameters is required to incorporate changes, i.e.,

```python
import pybamm

model = pybamm.lithium_ion.DFN()
sim = pybamm.Simulation(model)

sim.model = pybamm.lithium_ion.DFN()   [... cut]