pybamm-team/PyBaMM #3197: Add documentation for Python scripts in `examples/` in accordance with `sphinx-gallery`

agriyakhetarpal, 2023-07-27:
We use `sphinx-gallery` for rendering thumbnail galleries in the built docs for the Jupyter notebooks, which are in the `docs/source/examples/` folder. This issue describes tasks to add the example scripts too with said extension. This will require the following:

- [ ]  Moving the example scripts in the `examples/` folder to `docs/source/examples/`
- [ ]  Adding documentation in docstrings](https://sphinx-gallery.github.io/stable/syntax.html#embedding-rst), intertwined with code with the contents detailed in ReST format
- [ ]  Rendering the example scripts with Sphinx and related configuration (adding downloads, making sure the thumbnail galleries are rendered correctly)
- [ ]  Fixing any broken links that occur while restructuring the toctrees.

Ideally, tasks 2 and 3 can be completed in a single PR that follows the completion of the first task

agriyakhetarpal, 2023-09-03:
Hi, I wouldn't recommend using language models for this task; their outputs might put you in a position of debug hell. The documentation for `sphinx-gallery` and its working details are fairly straightforward

agriyakhetarpal, 2023-12-14:
Hello @iapoursanidis, how is this coming along? Is there some progress you wish to share with us?

Bashamega, 2024-05-07:
I can complete task 1

agriyakhetarpal, 2024-05-07:
Hi, @Bashamega, we would prefer that someone takes all four tasks (or at least a significant section out of them) in the same PR. This issue can be time-taking to implement. This is because moving the scripts to `docs/source/examples/` (first task) will not serve much use without them being added to the documentation (second task), and therefore the tasks are inherently connected in their proposition.

Bashamega, 2024-05-07:
I can try to finish all the tasks in one go

rtimms, 2024-05-10:
Is there some way we can achieve this whilst having both scripts and notebooks living in `examples/` rather than `docs/source/examples/`? It's always surprising to me that they live so deep in docs. I'm not sure how people currently access our examples, but it's confusing that they live in two separate places, and you would normally expect to find them directly in `example/`.

agriyakhetarpal, 2024-05-11:
@rtimms, the major issue is that Sphinx does not allow referencing items in toctrees from files external to the top-level directory where `conf.py` exists. One solution is to copy the scripts and notebooks at the time of the documentation build using a minimal Sphinx extension I can conjure up. Another cool method I've been recently thinking of using is this project: https://pypi.org/project/sphinx-collections/

@Bashamega, sorry, I just saw your response now. I would say that some of the scripts might need some knowledge of battery modelling for comments to be added, but surely not all of them. There are quite many scripts, so doing this can be a bit tedious as well. Please feel free to work on this if you wish to.