pybamm-team/PyBaMM #3219: Remove particle size parameters: mean, min, max, standard deviation

tinosulzer, 2023-08-01:
They are defined in the `GeometricParameters` and `LithiumIonParameters`
https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/geometric_parameters.py#L113-L115
but not used by any of the models

Having them in is a bit confusing, since changing them does not affect the distribution, which is entirely controlled by `f_a_dist` https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/size_distribution_parameters.py#L77

A user might think that they can change the distribution by changing one of the min/max/mean/sd parameters, but they will have no effect

tinosulzer, 2023-08-01:
Edit: `R_min` and `R_max` determine the geometry so should be kept

Very easy fix: remove the mean/sd parameters from `size_distribution_parameters.py`, and remove `.sd_a` in the (very few) places it appears. `R_mean` is never defined.

jeromtom, 2023-08-02:
I don't think we can remove `sd_p` and `sd_n` as they are used in the calculations of `R_min` and `R_max` which like you said determine the geometry.

Jay-sanjay, 2023-08-06:
@jeromtom  are you still working on this issue

jeromtom, 2023-08-06:
Yes

PR #3226 (2023-08-07): Removed sd_a- particle size parameters: mean, standard deviation
# Description
Problem:
Particle size parameters like mean/standard deviation are defined but not used, potentially leading to a situation where the user changes it but it will have no effect

Considerations:
-There isn't one mean standard deviation parameter in [size_distribution_parameter.py](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/size_distribution_parameters.py) but sd_p and sd_n for the area-weighted standard deviation of positive and negative electrodes respectively.
-`sd_p` and `sd_n` cannot be  [... cut]
closed 2023-08-07 by tinosulzer