pybamm-team/PyBaMM #3240: Failing readthedocs build

Saransh-cpp, 2023-08-05:
The [build is failing](https://readthedocs.org/projects/pybamm/builds/21517100/) with the following error -
```
Build output directory for format "pdf" contains multiple files and it is not currently supported.
Please, remove all the files but the "pdf" you want to upload.
```
Related to https://github.com/pybamm-team/PyBaMM/pull/3182?

cc: @agriyakhetarpal

agriyakhetarpal, 2023-08-06:
I haven't built or used documentation PDFs before so I gave it a try just now locally—I can see a bunch of individual PDF files generated for our custom inheritance diagrams for models and submodels, and a lot of PNG files for images generated in the notebooks by `nbsphinx` that are referenced in the `PyBaMM.tex` document, which I assume is then converted to PDF format with Sphinx's `LaTeXBuilder`. I don't know how to debug the errors and can't find many resources online about it, so should we discard the PDF outputs for the time being to fix the builds? (The EPUB format builds are working, at least). @tinosulzer, what do you think?

agriyakhetarpal, 2023-08-06:
We could use [rinohtype](https://github.com/brechtm/rinohtype) to build PDFs for our documentation shall we need them, it is an alternative that Sphinx itself recommends

tinosulzer, 2023-08-07:
Reopening until we find a permanent fix

Saransh-cpp, 2023-08-19:
Reopening as the build is failing again with the same error - https://readthedocs.org/projects/pybamm/builds/21656389/

agriyakhetarpal, 2023-08-20:
Not to worry, we had got to know Read the Docs uses some sort of caching and keeps older versions of builds for a while on our `develop` branch, while PRs always track the latest changes (except when there are changes to the content, which show up as usual). This was when I reported strange behaviour in the meeting three weeks ago about the version switcher continuing to display on docs.pybamm.org while it did not do so on PR builds, after its removal from the codebase entirely. It was resolved since it automatically disappeared by the time of the meeting after that

I think this is the same issue. Note that on any PR build for recent PRs – say, on #3276, the PST search bar overlay will not display upon pressing <kbd>⌘</kbd>+<kbd>K</kbd>, which is the intended behaviour because I changed the CSS attributes around it to hide it; it still shows up behind the Algolia search box on the main build. I suspect the failing build error should be gone within a week or two when the changes to the Sphinx extension from #3260 get reflected. Therefore we can close this issue safely—the PDF generation can be verified locally using the `sphinx-build -m latexpdf docs/ docs/build/` command: it generated multiple PDF files in the `docs/build/latex/` folder earlier to account for the SVG files for inheritance diagrams, but the `.. only:: not latex` directive now prevents it from doing so, as is the case with other projects that use the `sphinx.ext.inheritance_diagram` extension.

Saransh-cpp, 2023-08-20:
Oh I see, thanks for the detailed explanation and the examples, @agriyakhetarpal! :)

agriyakhetarpal, 2023-08-23:
Could we reopen this while I write a fix? Just for the sake of visibility

PR #3245 (2023-08-07): Temporarily disable PDF documentation builds
# Description

Removes PDF builds from the Read the Docs configuration, closes #3240

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist [... cut]
PR #3260 (2023-08-10): Re-enable PDF builds for Read the Docs
# Description

The SVG images generated by the `sphinxext/inheritance_diagram.py` extension create multiple PDFs upon LaTeX output with Sphinx's LaTeXBuilder. This PR makes them render on HTML output only, which is in-line with documentation for other projects that display inheritance diagrams for API components.

Closes #3240 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, als [... cut]
closed 2023-08-29 by Saransh-cpp