pybamm-team/PyBaMM #3250: [Bug]: Cannot access graphite_ocp_Enertech_Ai2020 data

michelegalasso, 2023-08-08:
### PyBaMM Version

23.5

### Python Version

3.10.12

### Describe the bug

I am trying to use the OCP curve of graphite as implemented in the "Chen2020_composite" parameter dataset, but when I try to access the data I get a TypeError. OCP curves of silicon and NMC implemented in the same dataset work as expected.

### Steps to Reproduce

    # load parameter values
    parameter_values = pybamm.ParameterValues('Chen2020_composite')

    # extract the OCP curves
    graphite_ocp_Enertech_Ai2020 = parameter_values['Primary: Negative electrode OCP [V]']

    # try to access the data (ERROR)
    ocp_value = graphite_ocp_Enertech_Ai2020(0.2)

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/michele/Documents/learning/python/PyBaMM/main.py", line 15, in <module>
    ocp_value = graphite_ocp_Enertech_Ai2020(0.2)
  File "/home/michele/Documents/learning/python/PyBaMM/.venv/lib/python3.10/site-packages/pybamm/input/parameters/lithium_ion/Chen2020_composite.py", line 318, in graphite_ocp_Enertech_Ai2020
    return pybamm.Interpolant(x, y, sto, name=name, interpolator="cubic")
  File "/home/michele/Documents/learning/python/PyBaMM/.venv/lib/python3.10/site-packages/pybamm/expression_tree/interpolant.py", line 116, in __init__
    if len(x) != len(children):
TypeError: object of type 'float' has no len()
```

brosaplanella, 2023-08-08:
Does this work?
```python3
ocp_value = graphite_ocp_Enertech_Ai2020([0.2])
```

michelegalasso, 2023-08-08:
Yes, this returns an object <class 'pybamm.expression_tree.interpolant.Interpolant'>, with child equal to 0.2.
However, I am not sure what does all that mean and how to obtain a vector of OCP in volts corresponding to different lithiation states of graphite.

tinosulzer, 2023-08-08:
You need to do something like `graphite_ocp_Enertech_AI2020(pybamm.Symbol(0.2)).evaluate()` to get a single value or `graphite_ocp_Enertech_AI2020(pybamm.linspace(0,1)).evaluate()` to get a vector

michelegalasso, 2023-08-08:
Now it works, thank you!! I had no idea that I needed to use pybamm.linspace(). Is that documented anywhere?

brosaplanella, 2023-08-10:
The linspace is documented in the docs (https://docs.pybamm.org/en/latest/source/api/expression_tree/array.html#pybamm.linspace) but there is no example of this. Transferring to a discussion as it might be useful to other users.