pybamm-team/PyBaMM #3274: testing runner with GPU support

martinjrobins, 2023-08-16:
### Description

Add test runner with GPU support to test gpu-specific code

### Motivation

https://github.com/pybamm-team/PyBaMM/pull/3121 introduced some GPU-specific code, which we need to test.

### Possible Implementation

We're using a mac mini for M1 testing and deployments, we could use this

### Additional context

@BradyPlanden

BradyPlanden, 2023-09-27:
Currently, JAX on Apple M-series GPU's is experimental and requires a build from source (see [here](https://developer.apple.com/metal/jax/)) with the a minimum JAX version of 4.11. Is there a reason PyBaMM pin's JAX to 4.7 & 4.8? As similarly discussed in #3371, it appears that we can relax this dependancy?

jsbrittain, 2023-09-27:
@BradyPlanden As far as I know we should be able to relax this requirement, at least to the minor version, i.e. 0.4 (there were some substantial API changes between 0.3 and 0.4).

agriyakhetarpal, 2023-09-27:
If it helps, I built `jaxlib==0.4.11` from source on an M2 Air from https://developer.apple.com/metal/jax/ as mentioned above. I then installed `jax==0.4.10` and `jax-metal==0.0.3` (i.e., the ones that do not cause version conflicts currently) and ran the unit tests with `python run-tests.py --unit`

<details><summary>Verifying that Jax uses the GPU</summary>
<p>

the output of

```python
import jax
jax.default_backend()
```

is

```
2023-09-27 23:54:55.469762: W pjrt_plugin/src/mps_client.cc:535] WARNING: JAX Apple GPU support is experimental and not all JAX functionality is correctly supported!
Metal device set to: Apple M2

systemMemory: 8.00 GB
maxCacheSize: 2.67 GB

'METAL'
```

</p>
</details>

<details><summary>Error logs</summary>
<p>

Note: the unit tests did not complete and exited early with this warning: `UserWarning: resource_tracker: There appear to be 1 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '`

Most Jax-related tests fail, all with the same error, e.g.,
```
test_evaluator_jax_jvp (test_expression_tree.test_operations.test_evaluate_python.TestEvaluate.test_evaluator_jax_jvp) ... "builtin.module"() ({
  "func.func"() ({
  ^bb0(%arg0: tensor<1x1xf64>):
    %0 = "mps.constant"() {value = dense<2.000000e+00> : tensor<1x1xf64>} : () -> tensor<1x1xf64>
    %1 = "mps.constant"() {value = dense<1.000000e+00> : tensor<1x1xf64>} : () -> tensor<1x1xf64>
    %2 = "mps.power"(%arg0, %1) : (tensor<1x1xf64>, tensor<1x1xf64>) -> tensor<1x1xf64>
    %3 = "mps.multiply"(%2, %0) : (tensor<1x1xf64>, tensor<1x1xf64>) -> tensor<1x1xf64>
    %4 = "mps.constant"() {value = dense<1> : tensor<4xsi64>} : () -> tensor<4xsi64>
    %5 = "mps.reshape"(%3, %4) : (tensor<1x1xf64>, tensor<4xsi64>) -> tensor<1x1x1x1xf64>
    "func.return"(%5) : (tensor<1x1x1x1xf64>) -> ()
  }) {arg_attrs = [{jax.arg_info = "y", mhlo.sharding = "{replicated}"}], function_type = (tensor<1x1xf64>) -> tensor<1x1x1x1xf64>, res_attrs = [{jax.result_info = ""}], sym_name = "main", sym_visibility = "public"} : () -> ()
}) {mhlo.num_partitions = 1 : i32, mhlo.num_replicas = 1 : i32, sym_name = "jit_evaluate_jax"} : () -> ()
ERROR
```

</p>
</details> 

From a quick look, https://github.com/google/jax/issues/16435 seems to be related. With the instability of `jax-metal` in its current state, perhaps it would be best to test this GPU-specific code on either Windows + WSL or an Ubuntu machine? On a GPU-enabled self-hosted runner, we could use https://github.com/marketplace/actions/cuda-toolkit for drivers. It looks like GPU-enabled runners are on GitHub's roadmap too (https://github.com/github/roadmap/issues/505).

BradyPlanden, 2023-09-28:
Right so a bit of investigation confirmed that M-series GPU's don't support `FP64`. In this case, `config.update("jax_enable_x64", True)` needs to be skipped in [evaluate_python.py](https://github.com/pybamm-team/PyBaMM/blob/88cf418b6bbee9d74c3faff8ab17f81fa899ed53/pybamm/expression_tree/operations/evaluate_python.py#L16C41-L16C41) and [jax_bdf_solver.py](https://github.com/pybamm-team/PyBaMM/blob/88cf418b6bbee9d74c3faff8ab17f81fa899ed53/pybamm/solvers/jax_bdf_solver.py#L21). Once that's done, the python `run-tests.py --unit` proceeds, but fail due to assertion tolerances (Jax is FP32 as default).

To proceed we could add os.platform logic to the `config.update("jax_enable_x64", True)`, in addition with GPU or CPU flags. Thoughts?

PR #3380 (2023-10-02): Jax on Metal w/ GPU support
# Description

This adds:
-  `FP32` support for Jax on Metal with GPU compilation (M-series GPU's don't support `FP64`)
-   Logic for parallelised GPU implementation on Metal

This fixes #3274; however, there are errors in the Jax GPU tests when ran on apple-silicon. I'll open a seperate issue for those though.

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breakin [... cut]
closed 2023-10-02 by BradyPlanden