pybamm-team/PyBaMM #3299: `AttributeError` in `examples/scripts/create-model.py`

Saransh-cpp, 2023-08-29:
But with `python examples/scripts/create-model.py` I am getting some attribute errors locally i.e. 

```
Traceback (most recent call last):
  File "/mnt/d/Arjun/PyBaMM/examples/scripts/create_model.py", line 99, in <module>
    param.process_geometry(geometry)
  File "/mnt/c/Users/Arjun/Desktop/PyBaMM/pybamm/parameters/parameter_values.py", line 463, in process_geometry
    for lim, sym in spatial_limits.items():
AttributeError: 'Scalar' object has no attribute 'items'
```

_Originally posted by @arjxn-py in https://github.com/pybamm-team/PyBaMM/pull/3235#discussion_r1301559059_

agriyakhetarpal, 2023-08-31:
I could not reproduce this on macOS with the latest changes from develop, it works fine for me

Saransh-cpp, 2023-08-31:
Oh yes, worked for me too. @arjxn-py you might be using a different version of PyBaMM. Closing this.

closed 2023-08-31 by Saransh-cpp