pybamm-team/PyBaMM #3303: Better option checking

rtimms, 2023-08-31:
The options checking doesn't always work as expected when options are tuples. For example, the following code performs checks if the option "particle size" is "distribution", but "particle size" can be a tuple, in which case the test doesn't run

```python3
        if options["particle size"] == "distribution":
            if options["lithium plating"] != "none":
                raise NotImplementedError(
                    "Lithium plating submodels do not yet support particle-size "
                    "distributions."
                )
```

The options should be converted to tuples first (where appropriate) and then checks performed on each index.

Saswatsusmoy, 2023-09-15:
Can i get the path for the same code?

Saswatsusmoy, 2023-09-16:
@rtimms Can you please review this and let me know if any more changes are required in the code.

rtimms, 2026-10-20:
This will be closed by #5816. Tuple-valued options now go through one resolver, and the particle-size-distribution checks (and the particle mechanics submodels) read `"particle size"` per electrode. A distribution on only one electrode is now caught, and a polynomial profile is only rejected on the electrode with the distribution. If you hit a whole-value check that still misses a tuple, please open a new issue.