pybamm-team/PyBaMM #3304: Pin versions

brosaplanella, 2023-08-31:
At the moment we do not pin to specific versions (in some cases we provide lower bounds) but this has the downside that installing old versions do not work (e.g. version 22.12 requires `casadi >= 3.5.0` which at the time meant 3.5, but now it would install 3.6 which is not compatible).

I don't know what's the right approach here, but just wanted to bring it up to discuss if we are happy with it as is, or if we want to be more specific with the pins (and if so, how).

agriyakhetarpal, 2023-08-31:
I read this blog recently about the nuances of versioning dependencies and learned a lot, it could offer some insights: https://iscinumpy.dev/post/bound-version-constraints/

Saransh-cpp, 2023-08-31:
TL;DR:

Yes, we should not add upper-bound version constraints unless they are extremely required (outdated PyBaMM versions should be allowed to break). Adding upper-bound constraints will make PyBaMM fall into anti-library patterns, converting it into an application and making it very hard to use it as a library (or as a dependency for other downstream libraries).

On the other hand, adding lower-bound pins is a good idea. We should definitely do that.

PS: The blog's author was my supervisor last year (learned all the python packaging and infrastructure stuff from him!) and we had a good amount of discussion on this. Upper-bound pins can really be a pain for downstream developers (some Scikit-HEP libraries were indeed facing a lot of difficulties from TensorFlow's upper-bound pinning policies).

brosaplanella, 2023-09-01:
Cool, then we are doing the right thing, right? Will close this.

closed 2023-09-01 by brosaplanella