pybamm-team/PyBaMM #3309: `python run-tests.py --doctest` (`nox -s doctests`) fails in CI

Saransh-cpp, 2023-09-05:
The doctest step is failing in every workflow. @agriyakhetarpal I think the breaking point is #3292. Are any updates required in the `noxfile` to sync RTD and GH Actions builds?

agriyakhetarpal, 2023-09-05:
I don't think it is unrelated to my changes, we identified this problem here: https://github.com/pybamm-team/PyBaMM/pull/3242#issuecomment-1700862027

arjxn-py, 2023-09-05:
I'm also looking into the same.

Saransh-cpp, 2023-09-05:
Oh oops. That makes sense. Thanks!

arjxn-py, 2023-09-05:
Possibly there might be dependency conflicts as when I am running doctests locally - 
```
Exception occurred:
  File "/home/arjxnpy/miniconda3/envs/py39/lib/python3.9/site-packages/pkg_resources/__init__.py", line 875, in _resolve_dist        
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (docutils 0.19 (/home/arjxnpy/miniconda3/envs/py39/lib/python3.9/site-packages), Requirement.parse('docutils!=0.18.*,!=0.19.*,>=0.8'))
```

Then when I am installing `docutils` using pip
```
Collecting docutils==0.8
  Using cached docutils-0.8.tar.gz (1.5 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [4 lines of output]
      Error: The "distutils" standard module, which is required for the
      installation of Docutils, could not be found.  You may need to
      install a package called "python-devel" (or similar) on your
      system using your package manager.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

agriyakhetarpal, 2023-09-05:
I tried it with Python 3.9 on macOS but could not reproduce it, it installs `docutils==0.18.1` for me, rather than 0.19. @arjxn-py arjxn-py could you try uninstalling `docutils` and purging the pip cache before retrying the doctests? It might be an issue with how `distutils` is getting removed everywhere since it is deprecated. Also, you seem to be installing `docutils==0.8` in the second error log, which is a very old version

PR #3311 (2023-09-06): Issue 3309 Fix doctests
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #3309 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change  [... cut]
closed 2023-09-06 by Saransh-cpp