pybamm-team/PyBaMM #3319: MacOS Nox documentation build error

kratman, 2023-09-11:
While trying to build the documentation, I ran into an issue.

Running ```nox -s docs``` produces the following error:
```
Session docs raised exception AttributeError('__enter__')
```

The nox session for docs is the following:
```python
@nox.session(name="docs")
def build_docs(session):
    """Build the documentation and load it in a browser tab, rebuilding on changes."""
    envbindir = session.bin
    session.install("-e", ".[all,docs]")
    with session.chdir("docs/"):
        session.run(
            "sphinx-autobuild",
            "-j",
            "auto",
            "--open-browser",
            "-qT",
            ".",
            f"{envbindir}/../tmp/html",
        )
```
The AttributeError is caused by the "with" clause in the code above.

The error no longer occurs when I remove the "with" block as shown below:
```python
@nox.session(name="docs")
def build_docs(session):
    """Build the documentation and load it in a browser tab, rebuilding on changes."""
    envbindir = session.bin
    session.install("-e", ".[all,docs]")
    session.chdir("docs")
    session.run(
        "sphinx-autobuild",
        "-j",
        "auto",
        "--open-browser",
        "-qT",
        ".",
        f"{envbindir}/../tmp/html",
    )
```

Has anyone else encountered this problem? I have my change on a branch and can make a pull request. I am guessing it is an issue with the version of Nox I have installed. However, if this small change could help other users, then is there a strong reason not to do it? 

I attached some basic system information below, I can add more if required.

System information:
MacOS
Apple silicon (M1 Chip)
Python 3.9
PyBaMM 23.5
Nox 2021.10.1

agriyakhetarpal, 2023-09-11:
Thanks for the MWE, I am able to reproduce the error—could you please update `nox` to a newer version (i.e., `nox==2023.4.22`)? The documentation builds successfully for me then.

agriyakhetarpal, 2023-09-11:
The error comes from the older version of `nox` not having a context manager at its disposal configured with `session.chdir`syntax, which is present in the latest version and thus enables the use of the `with` keyword

kratman, 2023-09-11:
Yes I will update my version of Nox and I am going to try to figure out why such an old version was installed for my Mac. The current version on Conda is 2023.4.22, but installing it this week from conda-forge downloaded 2021.10.1. Likely something from my initial environment setup.

Is there an interest in switching away from the "with" keyword so it is backwards compatible, or should I discard my branch?

agriyakhetarpal, 2023-09-11:
It could be a nice thing to make it backwards-compatible: @Saransh-cpp, what do you think? For now, you could use the `sphinx-autobuild` command that it wraps to get the build working

kratman, 2023-09-11:
Ok I will make the PR and then it can be closed if the change is not wanted.

kratman, 2023-09-11:
@agriyakhetarpal Change done in #3322

PR #3322 (2023-09-12): Remove with clause in docs session for backwards compatibility
# Description

The "with" keyword in the ```nox -s docs``` session causes an AttributeError in older versions of Nox. This change makes the nox session backwards compatible.

Fixes #3319

## Type of change

Closest category is "bug fix", however, this is just a minor compatibility issue.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-a [... cut]
closed 2023-09-12 by Saransh-cpp