pybamm-team/PyBaMM #3323: [Bug]: Unable to obtain the right results of "{Domain} electrode thickness change [m]".

bobonice, 2023-09-12:
### PyBaMM Version

v23.5

### Python Version

3.11

### Describe the bug

## Unable to obtain the right results of "{Domain} electrode thickness change [m]". 
For example, in the tutorial ["Validating_mechanical_models_Enertech_DFN.ipynb”](https://github.com/pybamm-team/PyBaMM/blob/develop/docs/source/examples/notebooks/models/Validating_mechanical_models_Enertech_DFN.ipynb) , the simulation and experimental results of 'thickness change [m]' in Figure 1 do not match at all. 

[image]

It could be caused by the absence of dimension for the variable "**electrode_thickness_change**" in the [class BaseMechanics (base_mechanics.py)](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/particle_mechanics/base_mechanics.py). In reference to the previous version, the bug could be fixed using the following code:

- change the code on line 58 to:
`L0 = domain_param.L`
`electrode_thickness_change = self.param.n_electrodes_parallel * v_change * L0`
**or**
- change the lines 96-97 to:
`neg_thickness_change = variables["Negative electrode thickness change [m]"] * self.param.n.L`
`pos_thickness_change = variables["Positive electrode thickness change [m]"] * self.param.p.L`

After applying the above change, the new Figure 1 in the tutorial "Validating_mechanical_models_Enertech_DFN.ipynb” would be like this:
[image]

### Steps to Reproduce

Perform the simulations which include the absolute thickness changes or run the code directly in the tutorial "Validating_mechanical_models_Enertech_DFN.ipynb”.

rtimms, 2023-09-12:
Good catch, thanks. I prefer the first solution. Could you open a PR with the fix?

bobonice, 2023-09-13:
Thank you for your prompt response and feedback. I have opened the [PR#3329.](https://github.com/pybamm-team/PyBaMM/pull/3329)

brosaplanella, 2023-09-15:
This was fixed in #3329, so closing this issue

closed 2023-09-15 by brosaplanella