pybamm-team/PyBaMM #3328: Released version of PyBaMM does not match tutorials

kratman, 2023-09-13:
While working on the tutorials, I noticed that tutorial 4 was failing when I downloaded it from the user guide.

Cell:
```python
parameter_values["Current function [A]"] = 10
parameter_values["Open-circuit voltage at 100% SOC [V]"]=3.4
parameter_values["Open-circuit voltage at 0% SOC [V]"]=3.0
```
Result:
```bash
KeyError: "'Open-circuit voltage at 100% SOC [V]' not found. Best matches are ['SEI open-circuit potential [V]', 'Outer SEI open-circuit potential [V]', 'Upper voltage cut-off [V]']"
```

After digging into it a bit I noticed that the released version from both pip and conda lack the recently renamed "Open-circuit voltage at XX% SOC [V]" parameters from Chen2020.
```bash
(envPyBaMM) kratz@Kratzs-MacBook-Pro ~ % python
Python 3.9.18 | packaged by conda-forge | (main, Aug 30 2023, 03:53:08) 
[Clang 15.0.7 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pybamm
>>> pybamm.__version__
'23.5'
>>> params = pybamm.ParameterValues("Chen2020")
>>> params.search("Open-circuit")
Inner SEI open-circuit potential [V]	0.1
Outer SEI open-circuit potential [V]	0.8
SEI open-circuit potential [V]	0.4
```
So the tutorials appear to have been updated before the new parameters were added to the release.

Is a new release of PyBaMM coming so that the tutorials will work with the pip/conda releases? The tutorials work fine when running in my IDE using PyBaMM installed from source.

agriyakhetarpal, 2023-09-13:
Thanks for reporting this bug @kratman, I have pushed a fix so that the links to the notebook downloads point to the correct version of the documentation at #3333. Could you please check if that fixes your issue? In any case, a new 23.9 release of PyBaMM will be coming soon on PyPI and conda, which will resolve the discrepancy

kratman, 2023-09-14:
@agriyakhetarpal Thanks I will take a look

PR #3333 (2023-09-14): Issue 3328 correctly point notebook download URLs according to versions
# Description

Closes #3328; ensures that users of the example notebooks can download the correct notebook from GitHub according to the latest version, stable versions starting from 23.5, and on PR builds. The Google Colab badge now points to the correct renditions of the notebooks, rather than the `develop` branch.

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR  [... cut]
closed 2023-09-14 by Saransh-cpp