pybamm-team/PyBaMM #3337: [Bug]: Using OKane2022 parameter set works with the K both upper and lower case

l-olisa, 2023-09-15:
### PyBaMM Version

23.5

### Python Version

3.11.5

### Describe the bug

When using params = pybamm.ParameterValues("OKane2022"), it works using both "Okane2022" and "OKane2022". For some group members the Okane2022 doesn't work (using same python and pybamm versions as above), and for others it does work (all using Windows).

### Steps to Reproduce

import pybamm

model_SEI = pybamm.lithium_ion.DFN(
     {"SEI": "solvent-diffusion limited",
     "SEI porosity change": "true", #When SEI is growing, porosity is changing. Ensure you add this, otherwise porosity (of electrolyte) does not change automatically (not porosity of SEI)
     })

params = pybamm.ParameterValues("OKane2022")

l-olisa, 2023-09-15:
Double checked and the PyBaMM versions were different for the laptops that didn't work (23.4.11). When upgraded to 23.5, able to run both upper and lower cases of OKane2022.

brosaplanella, 2023-09-15:
We followed this up and it was fixed in #3132 so closing this issue.

closed 2023-09-15 by brosaplanella