pybamm-team/PyBaMM #3339: [Bug]: Dead lithium not recorded in irreversible lithium plating model

tomjholland, 2023-09-15:
### PyBaMM Version

23.5

### Python Version

3.10.13

### Describe the bug

"Dead lithium" is recorded as a variable when the "partially reversible" lithium plating model is used, but not when the "irreversible" plating model is used. In the "irreversible" model, all plated lithium would be expected to become "dead lithium".

### Steps to Reproduce

Following code plots dead lithium concentration for both model types as it is formed:

```plating_models = ["partially reversible", "irreversible"]
params = pybamm.ParameterValues("OKane2022")
sol = []
plt.figure()
for plating_model in plating_models:
    model = pybamm.lithium_ion.DFN(options={"lithium plating": plating_model})
    sim = pybamm.Simulation(model, parameter_values = pybamm.ParameterValues("OKane2022"))
    sol=sim.solve([0, 3600])
    Qt = sol["Throughput capacity [A.h]"].entries
    Dead = sol["Dead lithium concentration [mol.m-3]"].entries[-1,:] # index at interface
    plt.plot(Qt,Dead,label=plating_model)
    plt.xlabel("Throughput capacity [A.h]")
    plt.ylabel("Dead lithium concentration [mol.m-3]")
    plt.legend()
plt.show()

rtimms, 2023-09-15:
cc @DrSOKane

DrSOKane, 2023-09-25:
Hi Tom, this is nothing to worry about. In the irreversible plating model, all plated lithium is dead, so there is no need for a separate dead lithium variable. Lithium is still conserved. That said, it will be an easy change to make once PR #3198 is complete.

brosaplanella, 2023-09-25:
I think it would be good to add the dead lithium variable for all models (even if it matches the plated lithium) just for clarity. Happy to wait until #3198 is merged

PR #3485 (2023-11-22): Issue 3339 dead lithium
# Description

The irreversible plating model now increases the `f"{Domain} dead lithium concentration [mol.m-3]"` variable, not the `f"{Domain} lithium plating concentration [mol.m-3]"` variable as it did previously. The `f"Loss of lithium to {domain} plating [mol]"` and `f"Loss of capacity to {domain} plating [A.h]"` variables are unchanged because they include both plated lithium and dead lithium.

Fixes #3339