pybamm-team/PyBaMM #3346: [Bug]: failing example notebook and script for COMSOL discharge curves

agriyakhetarpal, 2023-09-16:
### PyBaMM Version

develop

### Python Version

3.9

### Describe the bug

The example notebook `models/compare-comsol-discharge-curve.ipynb` and the script `examples/scripts/compare_comsol/discharge_curve.py` are failing in some workflow runs. 

Also, the example scripts were not being tested in the scheduled tests. I have opened a PR at #3345 to fix that

The links to the error logs are here: 

1. https://github.com/pybamm-team/PyBaMM/actions/runs/6204934533/job/16847574605#step:14:173
2. https://github.com/pybamm-team/PyBaMM/actions/runs/6208214736/job/16854166967?pr=3344#step:9:188
3. This is also noted in https://github.com/pybamm-team/PyBaMM/pull/3335#issuecomment-1721391033 and also while @arjxn-py and I were working on #3344.

### Steps to Reproduce

Run the example notebook and script with the latest changes from the `develop` branch

kratman, 2023-09-16:
I have a fix for discharge curve, and I am going to push it soon

agriyakhetarpal, 2023-09-16:
I see, I guess Matplotlib has a deprecation warning `MatplotlibDeprecationWarning: Auto-removal of overlapping axes is deprecated since 3.6 and will be removed two minor releases later; explicitly call ax.remove() as needed.` that might be related. I have 3.7.3 so this is what might have been removed in 3.8

kratman, 2023-09-16:
The failures started Friday morning. I saw it the CI for #3335. I was planning on digging into the solver issue this evening.
```
SolverError: Maximum number of decreased steps occurred at t=0.0 (final SolverError: 'Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_FIRST_RES_FAIL". Consult IDAS documentation.'). For a full solution try reducing dt_max (currently, dt_max=600) and/or reducing the size of the time steps or period of the experiment. Set `return_solution_if_failed_early=True` to return the solution object up to the point where failure occured.
```

kratman, 2023-09-16:
> I was planning on digging into the solver issue this evening.
>
> ```
> SolverError: Maximum number of decreased steps occurred at t=0.0 (final SolverError: 'Error in Function::call for 'F' [...]

@agriyakhetarpal Sorry for the confusion, I just found that the solver issue was due to a problem on my local setup. I do not see a solver failure in the logs from my CI runs. I think #3349 should be sufficient to close this issue (assuming CI passes).

agriyakhetarpal, 2023-09-16:
If you are on an M-series macOS machine then #3140 might be related

kratman, 2023-09-16:
Yeah I have an M1, they are a bit of a pain.

PR #3349 (2023-09-18): Bug fix: Replace prop_cycler with Cycler()
# Description

Matplotlib updated to 3.8 on Friday. Since then the line
```python
color = next(ax._get_lines.prop_cycler)["color"]
```
causes the following error:
```
AttributeError: '_process_plot_var_args' object has no attribute 'prop_cycler'
```

I have replaced the prop_cycler with a cycler object and configured it with a generic color order. This was the most minimal change I could think of to resolve the issue. I am open to suggestions for a specific color set or I can just remove the explicit cycler and let matplotlib handle the colors itself. [... cut]