pybamm-team/PyBaMM #3375: Decay rate for single state hysteresis models

tommaull, 2023-09-28:
### Description

Single state hysteresis models typically feature a decay rate. So far, this has not been implemented in pybamm as far as I can see. 

Previous hysteresis models implemented in pybamm: 
https://github.com/pybamm-team/PyBaMM/pull/2256 and https://github.com/pybamm-team/PyBaMM/pull/3194 

### Motivation

Voltage due to hysteresis can have a long timescale, which is left unmodeled in pybamm. This prevents non-instantaneous hysteresis dynamics from being modeled in pybamm. 

### Possible Implementation

Decay parameter, dependent on change in lithiation. Personally, I also think it would make more sense to have a single OCP with a hysteresis term than switch between two OCPs. 
[Latz_Plett_hysteresis (1).pdf](https://github.com/pybamm-team/PyBaMM/files/12748922/Latz_Plett_hysteresis.1.pdf)

ejfdickinson, 2023-09-28:
Is the recommendation that this applies to SPMe / DFN models with reference to the open-circuit potential and electrode reaction overpotential, for each electrode independently, and locally in the x-dimension?

The Latz and co-workers approach appears to be speaking of the full cell in terms of OCV and SOC. As a DFN / SPMe submodel alternative to `current sigmoid`, this would presumably need to work with OCP and lithiation extent respectively.

tommaull, 2023-09-28:
@ejfdickinson - yes. I hadn't thought about the x-dim as I usually work with SPMes, but that makes sense. I've opened a discussion here, as there's a number of improvements to OCP that could be made. https://github.com/pybamm-team/PyBaMM/discussions/3376

PR #3593 (2024-05-11): Plett OCP Hysteresis Model
# Description

Adds the hysteresis model as outlined in [Wycisk 2022](https://www.sciencedirect.com/science/article/abs/pii/S2352152X22010192?via%3Dihub), and as mentioned in 
#3375 and as discussed in #3376 .

This will allow for hysteresis switching to depend not on the current, but on the differential capacity change.  

Fixes #3375 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also a [... cut]
closed 2024-05-11 by valentinsulzer