pybamm-team/PyBaMM #3385: Move `install_odes` from `pybamm/` to `scripts/`

Saransh-cpp, 2023-10-01:
`install_odes.py` is a script and is not used by `pybamm` internally - https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/install_odes.py

akhilender-bongirwar, 2023-10-01:
@Saransh-cpp, I would like to work on this issue. Please assign me : ).

agriyakhetarpal, 2023-10-01:
We will also need to alter the entry point for `pybamm_install_odes` in this case, because the `scripts/` folder is not a module folder

Saransh-cpp, 2023-10-01:
I see.

Go ahead, @akhilender-bongirwar! The entry point should be edited here -

https://github.com/pybamm-team/PyBaMM/blob/7eb5879c264e26b02a71b0ced8ef11a5f092e451/setup.py#L293

akhilender-bongirwar, 2023-10-01:
> I see.
> 
> Go ahead, @akhilender-bongirwar! The entry point should be edited here -
> 
> https://github.com/pybamm-team/PyBaMM/blob/7eb5879c264e26b02a71b0ced8ef11a5f092e451/setup.py#L293

Hey @Saransh-cpp, I think I should change the entry point of this line 292 - `"pybamm_install_odes = pybamm.install_odes:main",` 

So should I remove this line from entry point as the `scripts/` is not a module folder.
Please correct me if I am wrong.

agriyakhetarpal, 2023-10-01:
Yes, that is correct, it should be the line above the one @Saransh-cpp quoted. However, we shouldn't remove the entry point altogether at this moment. Could you try adding an entry point to `install_odes` inside the `scripts/` folder (and investigate if that is possible)?

akhilender-bongirwar, 2023-10-01:
Yeah @agriyakhetarpal, I shall investigate.

akhilender-bongirwar, 2023-10-01:
@agriyakhetarpal, according to what I have investigated I am sure that entry point to `install_odes` can be added to `scripts/docker` file to `RUN if [ "$ALL" = "true" ]; then \` we can add at line 59 as `python scripts/install_odes.py && \` as the entry point. 

Please let me know if I am right about this :)

agriyakhetarpal, 2023-10-01:
We don't need to edit the Dockerfile in this case. This is related to PyBaMM's packaging infrastructure, the Docker installation picks it up as needed. `pybamm_install_odes` is an entry point, i.e., an app shipped with and enabled as such with the package, which can be run inside the root directory to install the build-time requirements (SUNDIALS) and an optional solver (`scikits.odes`)

akhilender-bongirwar, 2023-10-01:
@agriyakhetarpal, I understand. So, it seems there's no option to include 'install_odes' in the 'scripts/' folder then.

agriyakhetarpal, 2023-10-01:
@Saransh-cpp I am happy to close this since entry points [don't work outside importable module](https://packaging.python.org/en/latest/specifications/entry-points/#data-model) folders, unless you have some other ideas :) We shouldn't deprecate this command at the moment because it is useful for Linux after all.

Saransh-cpp, 2023-10-01:
Oh yes, it would be nice to make this script "uniform" with other scripts (like `install_KLU_Sundials.py`), that is, remove this as an entry point and encourage users to use `python scripts/install_odes.py`. If everyone thinks that keeping the script as an entry point is better than please feel free to close this!🙂

agriyakhetarpal, 2023-10-01:
A _hacky_ way around this would be to put the functionality in `scripts/install_odes.py`, switch to `pybamm.root_dir()`, and call it from the original `pybamm/install_odes.py` in a subprocess while capturing stdout and stderr in order to retain the `pybamm_install_odes` entry point. My preference would be towards keeping it as-is, though.

akhilender-bongirwar, 2023-10-02:
@Saransh-cpp, could you please summarize what exactly should I be doing to complete this issue and raise a PR.

arjxn-py, 2023-10-02:
> it would be nice to make this script "uniform" with other scripts (like `install_KLU_Sundials.py`), that is, remove this as an entry point and encourage users to use `python scripts/install_odes.py`.

This way we can encourage only developers to use `python scripts/install_odes.py`, removing `pybamm.install_odes:main` as an entrypoint might no longer leave `pybamm_install_odes` as user facing.

akhilender-bongirwar, 2023-10-02:
@arjxn-py , are you suggesting me to remove the entry point `pybamm.install_odes.main` ?

arjxn-py, 2023-10-03:
> are you suggesting me to remove the entry point `pybamm.install_odes.main` ?

No, i was suggesting to avoid removing this entrypoint as if we remove it then users might not have any way to access `python scripts/install_odes.py`

Saransh-cpp, 2023-10-03:
Oh yes, that is true, +1 on keeping things as it is. Closing this issue.

closed 2023-10-03 by Saransh-cpp

agriyakhetarpal, 2023-10-03:
Feel free to work on the other issue you commented on, @akhilender-bongirwar!