pybamm-team/PyBaMM #3386: Fix the URL checker

Saransh-cpp, 2023-10-01:
lychee is failing here at the moment -

https://github.com/pybamm-team/PyBaMM/blob/7eb5879c264e26b02a71b0ced8ef11a5f092e451/CONTRIBUTING.md?plain=1#L188

We could use a different material to reference glob pattern?

Agnik7, 2023-10-01:
Hi @Saransh-cpp I am participating in Hacktoberfest and would love to work on this issue!! Kindly assign this to me.

Saransh-cpp, 2023-10-02:
Hi, go ahead!

Agnik7, 2023-10-02:
@Saransh-cpp I have one query. I need to work and then send the PR to the `develop` branch right?

Saransh-cpp, 2023-10-02:
Yes. You can also refer to the [Contributing Guide](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md) for detailed steps.

Agnik7, 2023-10-02:
Ok thanks

Agnik7, 2023-10-02:
@Saransh-cpp I have created a PR #3393 . Requesting for your review. Thanks!!

PR #3393 (2023-10-08): fix: Fix URL Checker
# Description

Fixed the broken glob pattern link in CONTRIBUTING.md. I have replaced the previous link with the following [link](https://docs.python.org/3/library/glob.html)
Fixes #3386 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the  [... cut]
closed 2023-10-08 by Saransh-cpp