pybamm-team/PyBaMM #3387: Add a link checker step in the doctests jobs and fix broken links that appear

agriyakhetarpal, 2023-10-01:
Currently, we use Lychee to check for broken links, but we ignore a lot of Sphinx-related links because of referencing files internally. `Sphinx` provides a [`linkcheck`](https://www.sphinx-doc.org/en/master/usage/builders/index.html#sphinx.builders.linkcheck.CheckExternalLinksBuilder) builder for this purpose that can be invoked in CI for checking links (either through a new `nox` session, or within the `docs` session using posargs, or just calling the command directly).

It might make sense to rename the Lychee workflow to something like `links.yml` and add a parallel job to run the above command instead of another step.

RohitP2005, 2024-12-17:
hey @Saransh-cpp is the issue still open ?

agriyakhetarpal, 2024-12-17:
Hi @RohitP2005, this is still open, thanks for your interest!

RohitP2005, 2024-12-17:
Thanks, @agriyakhetarpal i will start my work

PR #4712 (2024-12-27): Updated readthedocs.yml to linkcheck
# Description

This PR introduces a new pre-build step to the `readthedocs.yml` configuration for PyBaMM. The change adds a `linkcheck` operation to ensure that all links in the documentation are checked for validity during the build process on Read the Docs.

### Changes made:
- Added a `pre_build` step to the `readthedocs.yml` file.
- The `pre_build` step runs the `sphinx -b linkcheck` command, which checks all links in the documentation before the build proceeds.
- The command is configured to have a short timeout (`linkcheck_timeout=1`) to handle any sl [... cut]
closed 2024-12-27 by kratman