pybamm-team/PyBaMM #3388: Make it possible to activate the venv in the `dev` `nox` session

agriyakhetarpal, 2023-10-01:
In `noxfile.py` we define a `dev` session that installs PyBaMM's dependencies for use by developers and contributors. It is possible to go one step further: automatically activate the virtual environment it creates with some logic based on the OS or the platform. See https://nox.thea.codes/en/stable/cookbook.html#instant-dev-environment for more information.

akhilender-bongirwar, 2023-10-01:
- According to my understanding, you want to integrate the code to create virtual environment from this  https://nox.thea.codes/en/stable/cookbook.html#instant-dev-environment  into `noxfile.py` in `dev` session.

@agriyakhetarpal , could you please explain this issue in detail if I am not correct about it.
Please assign me this issue : ).

agriyakhetarpal, 2023-10-01:
> * According to my understanding, you want to integrate the code to create virtual environment from this  https://nox.thea.codes/en/stable/cookbook.html#instant-dev-environment  into `noxfile.py` in `dev` session.

Yes. We already have a `dev` session in `noxfile.py` defined as follows:

https://github.com/pybamm-team/PyBaMM/blob/7eb5879c264e26b02a71b0ced8ef11a5f092e451/noxfile.py#L122-L138

It installs PyBaMM in editable mode with `pip` and sets environment variables accordingly. The goal is to modify the creation of this `nox` session so that it can not only install an editable rendition of PyBaMM inside a virtual environment with `virtualenv`, but also activate that environment at the completion of the session invocation. This way, a user should be able to run `nox -s dev` and subsequently a virtual environment folder (something like `venv/` or any of the other names present in the `.gitignore` file) would be created, and the environment can be activated. Some documentation about this change can be added afterwards.

Saransh-cpp, 2023-10-01:
@akhilender-bongirwar it is usually a good idea to work on one issue at a time when you're contributing to a new repository 🙂

akhilender-bongirwar, 2023-10-01:
@Saransh-cpp, I wanted to update you that the previous issue is nearly complete, and the only remaining aspect is determining the exact location to add the entry point for 'install_odes.py'. Therefore, I've shifted my focus to this new issue. Thank you for your input; I've noted it : )

akhilender-bongirwar, 2023-10-04:
Hello @Saransh-cpp, I wanted to mention that there's no problem at all. In fact, I found the community discussion to be quite beneficial. It provided me with valuable insights and knowledge related to the issue #3385.

PR #3408 (2023-10-18): fix: Implement Automatic Activation of Virtual Environment in the dev nox session
# Description

- Implement automatic activation of the virtual environment in the 'dev' Nox session
- Enhance the 'dev' Nox session defined in `noxfile.py`
- Simplify the development setup process for contributors and developers

This commit enhances the 'dev' Nox session to automatically activate the virtual environment when running 'nox -s dev'.

Fixes #3388

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to documen [... cut]
closed 2023-10-18 by Saransh-cpp