pybamm-team/PyBaMM #3389: Doctests do not remove built files if they fail

agriyakhetarpal, 2023-10-01:
In `run-tests.py`, if the doctests fail locally for some reason, they do not delete the built files and cause problems upon invoking the doctests again. This is because `nbsphinx` duplicates some of the doctrees upon rebuilds (it is not meant to work with the doctests). The command should be fixed in order to delete the built files upon invoking the doctests, or just the .doctrees folder inside them so that the error logs can be retained.

Agnik7, 2023-10-01:
Hi I would like to work on this issue. I am new to open-source, so I bit of guidance will be helpful.  Kindly assign it to me

agriyakhetarpal, 2023-10-01:
Hi @Agnik7, thanks for the help! For more explanation, we have a method in `run-tests.py` defined like this, which runs these tests through the command line (`python run-tests.py --doctest` or `nox -s doctests`)

https://github.com/pybamm-team/PyBaMM/blob/develop/run-tests.py#L61-L97

The issue is with the `subprocess.Popen` instance, because the process remains open until the terminal window is closed even if the doctests. Therefore, the built files (inside the `docs/build/` folder) are not deleted. We should alter it to use `subprocess.check_call` or `subprocess.run` with `check=True`, as appropriate, such that it fails on receiving a non-zero exit code (the doctests are supposed to treat warnings as errors) and closes the process. Try commenting out this line `shutil.rmtree("docs/build")` and run the doctests again to debug this issue. The ideal functionality is that we should remove the `.doctrees/` folder inside `docs/build/` regardless of whether the doctests fail or pass so that the doctrees can be re-built.

It would be helpful to set up the project locally using the source installation instructions based on your platform or inside a Docker container (see https://docs.pybamm.org/en/latest/source/user_guide/installation), and then you may try building the documentation and running the doctests by reading through the contributor guide. @Saransh-cpp, could you please assign them this?

shubhambhar007, 2023-12-05:
Hi @agriyakhetarpal @Saransh-cpp if no one is working on this could you please assign it to me ?Thanks :D

agriyakhetarpal, 2023-12-23:
Hi @shubhambhar007, how is this issue coming along?

shubhambhar007, 2023-12-23:
Hi @agriyakhetarpal ,apologies for the delay , was on call this month :(

Will raise the PR by tonight for review, max by tomorrow morning.

Thanks !!

agriyakhetarpal, 2023-12-23:
No worries, thanks for the update!

PR #3655 (2024-01-03): Doctests do not remove built files if they fail #3389
# Description

As pointed by @agriyakhetarpal , in case of failure of doctests, built files (inside the docs/build/ folder) are not deleted.
We also have to ensure deletion in case of successful doctests. I added a finally block for the same and put the subprocess call in a try block.

Fixes #3389

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the lis [... cut]
closed 2024-01-03 by kratman