pybamm-team/PyBaMM #3390: Drop support for Python `3.8`, support Python `3.12`?

agriyakhetarpal, 2023-10-01:
NumPy>=1.26 and SciPy>=1.11 support Python 3.9+ now and already have wheels for Python 3.12. The blocker should be CasADi, which might have a new release soon because the [pre-release](https://github.com/casadi/casadi/releases/tag/nightly-py312) is available. In any case, we should be able to drop Python `3.8` support at this time and `3.12` can wait

Saransh-cpp, 2023-10-01:
I don't think we should drop support for Python 3.8 unless PyBaMM breaks on 3.8 or 3.8 reaches `end-of-life`.

We can actually start testing PyBaMM on 3.12 in the CI. IIRC, `pybind11` has already migrated to Python 3.12, and we can fetch CasADi's nightly wheels in the 3.12 job.

agriyakhetarpal, 2023-10-01:
Yes, it would be good to try supporting Python 3.12, which comes tomorrow. `xarray` doesn't run CI against 3.12 yet, but they list their Python dependency as >=3.9. I think we should be fine with using the nightly wheels for CasADi, should we let the optional dependencies catch up later?

> I don't think we should drop support for Python 3.8 unless PyBaMM breaks on 3.8 or 3.8 reaches end-of-life.

That makes sense, I just checked and Python 3.8 EOL is in October 2024, so we have just about a year to remove it

Saransh-cpp, 2023-10-02:
We should only build and ship pybamm wheels once every dependency works, but testing pybamm (just with the required dependencies) in the CI should be good.

(release candidates of 3.12 have been available on GH Actions for quite some time now)

Saransh-cpp, 2023-10-31:
Copying from slack -

> https://scientific-python.org/specs/ - like PEPs but for the Scientific Python ecosystem. I think we should follow them to stay up-to-date with other libraries.
>
> The first SPEC suggests removing Python 3.9 support by the end of this year, but I’m not sure if we should do that :grimacing:

agriyakhetarpal, 2023-10-31:
There are a lot of interesting things available in the Scientific Python ecosystem, perhaps we should also try using [`sp-repo-review`](https://learn.scientific-python.org/development/guides/repo-review/)? I guess this would be better off as a separate issue, please let me know if I should open one

Saransh-cpp, 2023-10-31:
Some of the failing checks in repo review are intentional from our end (for example - we removed codespell, prettier, and isort as pre-commit hooks because they were doing more harm than good), and some of them are potential GSoC projects (for example - pybamm wide typing hints), but yes an issue to track repo review's suggestions would be nice. We could either implement the suggestions, discard them, or mark them as implemented.

PS: we weren't using repo review before because it was Scikit-HEP specific, but now they've moved it to Scientific Python, which is nice!

agriyakhetarpal, 2023-10-31:
Makes sense. I will run the tool locally and see what we are missing, I cannot get it to to work inside the browser at all for some reason. There has been some discussion about a PR for adding type hints to PyBaMM models after they have been serialised in #3397 (or they were probably for the expression tree: not sure) in the developer meeting yesterday

pipliggins, 2023-11-03:
> Makes sense. I will run the tool locally and see what we are missing, I cannot get it to to work inside the browser at all for some reason. There has been some discussion about a PR for adding type [...]

I've been going through and adding type hints for the expression trees yes (they're more of a by-product from trying various methods for serialisation) - I'll make a PR asap for those.

agriyakhetarpal, 2023-11-16:
`casadi` now supports Python 3.12 (https://github.com/casadi/casadi/releases/tag/3.6.4) which checks all four of our core dependencies. I installed all dependencies with Python 3.12 locally without issues, except for `[jax]` (which will now require bumping the versions) and `[odes]` (I see that `numpy` removed `distutils` with Python 3.12+ which broke the wheel building process – I can file an issue upstream requesting for a fix). 

@Saransh-cpp: should we try for Python 3.12? We can still keep Python 3.8 as discussed above. I was supposed to work on #3443, which I can then close in the same PR to avoid merge conflicts. In any case, it will be merged only after the stable release, so we have a lot of time to look into any failures that occur.

Saransh-cpp, 2023-11-16:
> should we try for Python 3.12? We can still keep Python 3.8 as discussed above.

Yes! We can add a note in the docs about optional dependencies not yet supported for Python 3.12.

> and [odes]

Ah, I remember the maintainers of scikits.odes were waiting for NumPy's migration to meson to stop using `numpy.distutils`, but the repository is inactive. See my old PR https://github.com/bmcage/odes/pull/137 for some discussion on this. Creating an issue there should be good.

PR #3961 (2024-04-03): Remove support for python 3.8
# Description

Removes support for python 3.8

Fixes #3390

## Type of change

This is a breaking change since it reduces support for a python version. On the plus side it reduces the number of CI jobs.

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to set this up to run automatically when committing locally, in just two lines of code)
- [x] All tests pass: `$ python run-tests.py --all` (or `$ n [... cut]
closed 2024-04-03 by kratman