pybamm-team/PyBaMM #3392: Improve documentation

brosaplanella, 2023-10-02:
There are quite a few features that are not in the docs, and which would be fairly minor fixes. Opening a list here so we can keep track of them as we go and fix them in batches. Feel free to add more in the comments, and we can update the list.

- [ ] `print_parameter_info` not documented in `base_model.py`
- [ ] Experiment steps docstrings do not mention that arrays can be passed as values for drive cycles.

ejfdickinson, 2023-10-02:
I'd widen the one relating to "experiment steps" to make clear that it feels like bad practice to give detailed documentation for a constructor `pybamm.step._Step()` that is not supposed to be called in user code. Just list the permitted `value` formats and `kwargs` that can be used in the `pybamm.step.[...]` options.

ejfdickinson, 2023-10-02:
`pybamm.ProcessedVariable` documents the public attribute `data` as "the same as `entries`" but doesn't document `entries`.

tinosulzer, 2023-10-04:
We could look into Google Season of Docs for this https://developers.google.com/season-of-docs/docs/timeline

Seems like there is a lot to be done to follow best practices, not requiring domain knowledge.
@ejfdickinson seems like you have lots of good suggestions, would you be available to mentor?

ejfdickinson, 2023-10-05:
@tinosulzer Yes in principle, but can you suggest what the time commitment and overall working period would be?

My suggestions are all provoked empirically by using the library for different tasks! I made a suggestion in DM to @brosaplanella which I'll copy here on how to identify areas where documentation is misaligned with expected use patterns.

> My recommendation for the best thing you can do is get a couple of students who are good at Python but have never used PyBaMM before, and have them reproduce a few modelling papers, especially those that don't use PyBaMM. Then just sit over their shoulders and work out what is clear and what is not.

ejfdickinson, 2024-12-05:
@valentinsulzer @martinjrobins @brosaplanella 

I'm not sure if this is the right issue, but I was just giving some thought to this.

I think a significant self-documentation challenge for models is that the general behaviour of `__str__()` and `to_equation()` for more complex symbols is to recursively expand all children. So while the source can construct models very neatly by acting on what are already complex variables, this is difficult to inspect at runtime without reference to the source.

A good example is to try the following inspection:

```python
model = pybamm.lithium_ion.DFN(
    options={"thermal": "lumped"},
)
print(model.rhs[model.variables["Volume-averaged cell temperature [K]"]])
```

The corresponding equation is completely straightforward to read when set at `models.submodels.thermal.lumped:88`, but if you call `print()` or `.to_equation()` on the corresponding RHS variable, you get a wall of text that is explicit but hard to parse.