pybamm-team/PyBaMM #3395: Jax solver mode: only include output variables

BradyPlanden, 2023-10-02:
### Description

This continues #2796 for the JaxSolver. 

### Motivation

For parameter estimation, it is (computationally) benefitical to store only the required output variables. 

This issue also aims to provide functionality for jit-compiled parameter estimation or optimisation workflows. At the moment this isn't possible since the creation of the `ProcessedVariable` dict is completed with Casadi. Further discussion here: https://github.com/pybop-team/PyBOP/pull/25

agriyakhetarpal, 2024-03-30:
Hi, @BradyPlanden – you had mentioned that #3658 should have closed this issue, but I think we forgot to do so. Should we close this now?

BradyPlanden, 2024-04-03:
Yep, marking closed.

closed 2024-04-03 by BradyPlanden