pybamm-team/PyBaMM #3404: Extend `pybamm_install_odes` to include support for macOS systems

agriyakhetarpal, 2023-10-03:
The `pybamm/install_odes.py` script, i.e., the `pybamm_install_odes` entry point currently installs the build-time requirements (SUNDIALS), sets relevant environment variables, and `scikits.odes` for use by the optional solver. Considering that users can also use `scikits.odes` on macOS by pointing to a SUNDIALS installation using the `SUNDIALS_INST` environment variable, we can add suitable modifications to this script to make it work on all UNIX-based systems (macOS and Linux).

Some reference can be used from the `install_KLU_Sundials.py` which currently works for both macOS and Linux. However, a difference to be noted is that `install_odes.py` does not compile the KLU module, whereas the aforementioned script does.

This will help simplify the installation of `scikits.odes` and enable doing so without the usage of a `nox` session, and can then be tested in CI.

agriyakhetarpal, 2023-10-04:
<details><summary>Installation logs</summary>
<p>

```
$ pybamm_install_odes       
/Users/agriyakhetarpal/Desktop/PyBaMM/pybamm/scikits_odes_setup.log
Adding /Users/agriyakhetarpal/.local/lib to LD_LIBRARY_PATH in /Users/agriyakhetarpal/Desktop/PyBaMM/venv/bin/activate
Collecting scikits.odes
  Downloading scikits.odes-2.7.0.tar.gz (931 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 931.8/931.8 kB 6.0 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: scipy in ./venv/lib/python3.11/site-packages (from scikits.odes) (1.11.3)
Requirement already satisfied: numpy<1.28.0,>=1.21.6 in ./venv/lib/python3.11/site-packages (from scipy->scikits.odes) (1.26.0)
Building wheels for collected packages: scikits.odes
  Building wheel for scikits.odes (pyproject.toml) ... done
  Created wheel for scikits.odes: filename=scikits.odes-2.7.0-cp311-cp311-macosx_13_0_arm64.whl size=885590 sha256=9f954addb89481d0c6c169143aaa0267b816dc008ec8f901d1890283ae6c9373
  Stored in directory: /Users/agriyakhetarpal/Library/Caches/pip/wheels/41/7b/df/a45628ef0daaa6d2c60461a092198e27eaa4f72d995ed524a4
Successfully built scikits.odes
Installing collected packages: scikits.odes
Successfully installed scikits.odes-2.7.0
```

</p>
</details> 

<details><summary>From scikits_odes_setup.log</summary>
<p>

```
2023-10-04 22:48:38,065 - scikits.odes setup - INFO - Starting scikits.odes setup
2023-10-04 22:48:38,065 - scikits.odes setup - INFO - Looking for sundials at /Users/agriyakhetarpal/.local
2023-10-04 22:48:38,065 - scikits.odes setup - INFO - Found sundials at /Users/agriyakhetarpal/.local
```

</p>
</details> 

Looks like the output was truncated because I already have a SUNDIALS installation from `nox -s pybamm-requires`. So I guess it should be possible (and easier than I expected) to modify it, we just need to build without KLU (i.e., `"-DENABLE_KLU=OFF"` in the `cmake_args`, which I suspect is the default behaviour, so it can be omitted as well. Another thing to handle would be finding OpenMP on macOS similar to `install_KLU_Sundials.py`. Do you want to take this?

agriyakhetarpal, 2023-10-04:
It should also be fine to build _with_ KLU and reuse an existing SUNDIALS installation like it did for my case, because `scikits.odes` does not provide an interface for it at all

arjxn-py, 2023-10-05:
> Do you want to take this?

Sure, happy to pick it up.

agriyakhetarpal, 2023-10-05:
Idea: enable KLU and add logic for finding OpenMP in order for users and developers to reuse the SUNDIALS files, though this would work only for source installations and not for `pip` installations (which come bundled with the dynamic libraries already)

PR #3417 (2024-01-06): Extend `pybamm_install_odes` to include support for macOS systems
# Description

Fixes #3404 
## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [x] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-comm [... cut]
closed 2024-01-06 by Saransh-cpp