pybamm-team/PyBaMM #3409: [Bug]: Could not find consistent states: Could not find acceptable solution: solver returned NaNs

Mrzhang-hub, 2023-10-05:
### PyBaMM Version

23.5

### Python Version

3.8

### Describe the bug

pybamm.expression_tree.exceptions.SolverError: Could not find acceptable solution: solver returned NaNs
During handling of the above exception, another exception occurred:
pybamm.expression_tree.exceptions.SolverError: Could not find consistent states: Could not find acceptable solution: solver returned NaNs

### Steps to Reproduce

```
model = pybamm.lithium_ion.DFN(options={"SEI": "ec reaction limited",
		"SEI film resistance": "distributed",
		"SEI porosity change": "true"})
param = pybamm.ParameterValues('Prada2013')
param.update(
	{
		# SEI model
		"EC diffusivity [m2.s-1]": 2e-19,
		"Inner SEI partial molar volume [m3.mol-1]": 0.162 / 1690,
		"Outer SEI partial molar volume [m3.mol-1]": 0.162 / 1690,
		"Ratio of lithium moles to SEI moles": 2,
		"SEI resistivity [Ohm.m]": 2e5,
		"Initial inner SEI thickness [m]": 0,
		"Initial outer SEI thickness [m]": 5e-9,
		"EC initial concentration in electrolyte [mol.m-3]": 4541,
		"SEI kinetic rate constant [m.s-1]": 1e-12,
		"SEI open-circuit potential [V]": 0,
		"SEI growth activation energy [J.mol-1]": 0,
		"SEI growth transfer coefficient": 0.5,
	}, check_already_exists=False)
param["Current function [A]"] = "[input]"
t_eval = np.linspace(0, 100, 100)
solver = pybamm.CasadiSolver(dt_max=100)  
sim = pybamm.Simulation(model, parameter_values=param,solver=solver)
sim.solve(initial_soc=0.1, t_eval=t_eval, inputs={"Current function [A]": 23})
sim.plot()
```

Mrzhang-hub, 2023-10-05:
I think the error maybe caused by the SEI model because the following code works fine.
```
model = pybamm.lithium_ion.DFN()
param = pybamm.ParameterValues('Prada2013')
param["Current function [A]"] = "[input]"
t_eval = np.linspace(0, 100, 100)
solver = pybamm.CasadiSolver(dt_max=100)   #dt_max默认为600，但是会报错，报错提醒让把dt_max调小
sim = pybamm.Simulation(model, parameter_values=param,solver=solver)
sim.solve(initial_soc=0.1, t_eval=t_eval, inputs={"Current function [A]": 23})
sim.plot()
```

brosaplanella, 2023-10-09:
This is probably also related to the parameters. The Prada2013 is not too reliable, some improvements have been done but they will only be available from v23.9 onwards only. If you want, you can check the updated parameters (see https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/input/parameters/lithium_ion/Prada2013.py).

Note as well that you start from a very low SOC and are doing a very high discharge rate, which might also cause issues.

kratman, 2024-05-03:
@Mrzhang-hub Is this still relevant?

closed 2024-06-03 by github-actions[bot]