pybamm-team/PyBaMM #3441: Call processed variables with custom spatial coordinates

rtimms, 2023-10-11:
After https://github.com/pybamm-team/PyBaMM/pull/3234 `ProcessedVariable` objects now automatically take the names of the domains to plot automatically from spatial variables, but `__call__` still expects the spatial variables to be `x`, `y`, `z`, `r`, or `R`, see:
```python3
    def __call__(self, t=None, x=None, r=None, y=None, z=None, R=None, warn=True):
        """
        Evaluate the variable at arbitrary *dimensional* t (and x, r, y, z and/or R),
        using interpolation
        """
        kwargs = {"t": t, "x": x, "r": r, "y": y, "z": z, "R": R}
        # Remove any None arguments
        kwargs = {key: value for key, value in kwargs.items() if value is not None}
        # Use xarray interpolation, return numpy array
        return self._xr_data_array.interp(**kwargs).values
```

We should make it possible to pass extra `kwargs` to `__call__`. 

As a workaround, you can do something like 
```python3
# variable that depends on a  spatial variable named "s" and time "t"
my_var_data_array = sim.solution["My var"]._xr_data_array

def my_var_eval(s, t):
    return my_var_data_array.interp(s=s, t=t).values
```

brosaplanella, 2023-10-11:
Yes! I thought of this when doing #3234 but didn't open an issue for it.

Saswatsusmoy, 2023-10-21:
Can we do something like this?

-Allow passing arbitrary keyword arguments to __call__ by adding **kwargs
-Get list of expected args from the method signature
-Get args passed to __call__ and kwargs
-Construct coord dict from expected args that were passed
-Update with any extra kwargs
-Pass coord dict to interp

rtimms, 2023-10-24:
@Saswatsusmoy sounds good, would you like to give it a go?

Saswatsusmoy, 2023-10-25:
@rtimms I have tried and modified the call method to accept **kwargs as a parameter. This will allow any additional keyword arguments to be passed to the method.

rtimms, 2024-11-22:
@valentinsulzer id this something that can be fixed as part of #4600 ?

arywk40-hue, 2026-11-16:
I'd like to work on this. `ProcessedVariable._process_spatial_variable_names` already keeps a custom name such as `s` as the coordinate, but `__call__` only accepts `x`, `r`, `y`, `z` and `R`. I plan to accept extra keyword arguments named after the variable's own spatial coordinates, raise an error for unknown names, and add tests.

Should an unknown keyword raise a `ValueError`, or would you prefer a `TypeError`, matching what Python raises for an unexpected argument?