pybamm-team/PyBaMM #3442: conda installation failure on Python 3.10 + updates required for v23.9

Saransh-cpp, 2023-10-11:
Hi! I believe PyBaMM now does support Python 3.10, but I'm still having trouble installing the latest version `23.5` using conda-forge.

I've tested on MacOS (x64) and Linux, and conda-forge tends to pick version `0.2.3.post1`.

Minimal example: `conda create -n pybamm-exp -c conda-forge python=3.10 pybamm=23.5`

Traceback:

```python-traceback
UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package python conflicts for:
python=3.10
pybamm=23.5 -> python[version='>=3.8,<3.9.0a0|>=3.9,<3.10.0a0']
pybamm=23.5 -> anytree[version='>=2.4.3'] -> python[version='*|2.7.*|3.5.*|3.6.*|>=3.6|>=3.10,<3.11.0a0|>=3.11,<3.12.0a0|>=3|>=3.7,<3.8.0a0|>=3.12,<3.13.0a0|>=3.6,<3.7.0a0|>=2.7,<2.8.0a0|>=3.12.0rc3,<3.13.0a0|>=3.7.1,<3.8.0a0|3.8.*|>=3.8|>=3.7|>=2.7|3.4.*|>=3.5,<3.6.0a0|>=3.9|>=3.5.3|>=3.5|3.9.*',build=*_cpython]The following specifications were found to be incompatible with your system:

  - feature:/osx-64::__osx==10.16=0
  - feature:/osx-64::__unix==0=0
  - feature:|@/osx-64::__osx==10.16=0
  - feature:|@/osx-64::__unix==0=0
  - pybamm=23.5 -> matplotlib-base[version='>=2.0'] -> __osx[version='>=10.11|>=10.12|>=10.9']
  - pybamm=23.5 -> sympy[version='>=1.8'] -> __unix

Your installed version is: 10.16
```

However, the following options work:
- python 3.9 & pybamm 23.5 with conda-forge
- python 3.10 & pybamm 23.5 with pip

_Originally posted by @pavithraes in https://github.com/pybamm-team/PyBaMM/discussions/2668#discussioncomment-7254827_

Saransh-cpp, 2023-10-11:
I think the Python versions should be updated here - https://github.com/conda-forge/pybamm-feedstock/blob/f1aa5221981530861780ec8a5e0a75da887178d1/recipe/meta.yaml#L18

agriyakhetarpal, 2023-10-11:
We should also add a section on verifying and/or updating the conda-forge package and/or recipe in the release reminders. For example, `pybamm_install_jax` still exists there but we use the optional dependencies for that now (note: not sure if we have completely removed the entry point or it is just in its deprecation phase).

priyanshuone6, 2023-10-15:
> I think the Python versions should be updated here - https://github.com/conda-forge/pybamm-feedstock/blob/f1aa5221981530861780ec8a5e0a75da887178d1/recipe/meta.yaml#L18

Yes, those selectors need updating
https://conda-forge.org/docs/maintainer/maintainer_faq.html#mfaq-py37-selector
https://docs.conda.io/projects/conda-build/en/stable/resources/define-metadata.html#preprocessing-selectors

agriyakhetarpal, 2023-10-30:
Hi @priyanshuone6, could you grant me access to the `pybamm-feedstock` repository? I can take a look at fixing those selectors before the release, plus I would love to learn a bit about how `conda` recipes are structured. The YAML looks straightforward enough to me, please let me know if you have any other pointers on what else is to be changed

priyanshuone6, 2023-10-31:
Done

closed 2023-11-09 by agriyakhetarpal